openscad/openscad #1957: Add a debugging mode for polyhedrons with point numbers

krichter722, 2017-02-22:
Since debugging polyhedrons sometimes causes beginners' brain cells to to vaporize (at least mine) it'd be nice to have a debugging mode which displayed the sequence counter of polyhedron points on the screen. In complex cases I only manage to fix `WARNING: Object may not be a valid 2-manifold and may need repair! ` by drawing the object on paper and add the point numbers - then it's astonishingly easy.

experienced with 2015.03-1

nophead, 2017-02-22:
I have seen people do that with their own code. They pass the list of
points to a module that numbers them with text() and translates it close to
the point. Other people use colour coded spheres.

kintel, 2017-02-22:
I agree that a utility module sounds like a good start. Making this a good built-in function is likely going to be a fun battle with corner cases : /

t-paul, 2019-11-21:
[BOSL](https://github.com/revarbat/BOSL) has a nice user space implementation. See https://github.com/revarbat/BOSL/wiki/debug.scad

closed 2019-11-21 by t-paul