openscad/openscad #1959: Feature request: script-level system resource usage query command.

royasutton, 2017-02-23:
New command that returns current system resource usage. The _getrusage_, get resource usage library, provides the need information. With such a command,  user-space libraries could be better optimized. The _getrusage_ library gives many use statistics. Would be nice to be able to specify the metrics of interested to be returned as a vector.

**For example**:
```
options = ["utime", "systime", " maxrss"];

ru0 = get_usage( options );
result1 = function1();
ru1 = get_usage( options );
result2 = function2();
ru2 = get_usage( options );

utime1 = ru1[0] - ru0[0];
utime2 = ru2[0] - ru1[0];

if ( utime1 < utime2 )
    echo("function1 is faster");
```

z3t0, 2017-03-25:
Would this be a function that can be called from within the model file?

kintel, 2017-03-26:
Once we start caching expression, you're need to clean caches between each measurement to get a true value.
My instinct says that this is better solved as a cmd-line wrapper, alternatively as an IDE helper.

royasutton, 2017-04-07:
My intention is with regards to library development and _dynamic code analysis_. I agree that one could do **coarse-grain** performance comparison from the command line. However, it becomes difficult to break things apart inside the code "box". At the command line level, it's easy to say box1 is better than box2. Which is fine in simple cases. But when one wants to optimize box1 or box2, a more **fine-grain** timing capability is needed.

For example,  if one was able to get time (or other resource) stamps from within a running script, one could begin to create performance profiles on sections of working code. This provides the key to in-the-box optimization.

I currently know nothing about the internals of OpenSCAD and with my 0-experience, it seemed like some low-hanging fruit that should not be too intrusive (it would only be used when needed for development, optimization, analysis, etc).

MichaelAtOz, 2017-04-07:
What about adding thread execution time to the multi-thread tree.
First step, dump that somewhere.
Later a GUI to view the tree next to code.
??

royasutton, 2017-04-08:
_Being able to gather the resource usages from within a running script is the key..._

Resources can fall into several groups (ie: '**man getrusage**' on linux/bsd). The OpenSCAD script writer could use the information that POSIX OS's already collects. What is missing is an OpenSCAD command that requests the current process resource usage from the OS, via getrusage(), and makes it available to the user in the *.scad script.

If such a command were added, the user could then develop fine-grain (ie: function/module level) profiling strategies as he/she feels fit within their scripts and even dump and post-process as desired to discover where optimizations are most warranted.

t-paul, 2017-04-08:
As there is no _running_ script as such, I think the idea from @MichaelAtOz is quite good and fits the descriptive model of OpenSCAD. Having some kind of tree structured display of the current script referenced to the source code would make lots of other neat features possible.

royasutton, 2017-04-08:
It just occurred to me that I overlooked an important point from @kintel. In order for the strategy to work, it would require an option to disable the cache. Yet another new command???

MichaelAtOz, 2017-04-09:
> I currently know nothing about the internals of OpenSCAD

Using Rocket example, source from [wiki](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Commented_Example_Projects#Rocket).

This is a simplified overview. Click to enlarge.
[image]
The compile is finished at the blue star.
After that it is graphics or geometric processing which is not directly related to specific bits of the source.
Compilation, apart from lots of deep recursion, takes a very small fraction of the total time.

My most complex, real world object, non-contrived, design, takes ~1s to compile [without render()'s], ~5s to preview & paint the window, & ~8m to render.

This is the normalised CSG where all processing effort goes, so you can't easily match the resource metrics to the source.

CSG after normalization:
((((((((hull34 - cylinder341) - cylinder345) - render346) - render820) - linear_extrude831) + (((cube41 - cylinder341) - render820) - linear_extrude831)) + ((((((((((((((cylinder51 * linear_extrude57) * cube63) - cylinder83) - cylinder85) - cylinder86) - cylinder87) - cylinder89) - cube161) + ((((((cylinder51 * linear_extrude57) * cylinder66) - cylinder86) - cylinder87) - cylinder89) - cube161)) + ((((((cylinder51 * cylinder70) * cylinder73) - cylinder86) - cylinder87) - cylinder89) - cube161)) + (((((((cylinder51 * cylinder77) * cylinder80) - cylinder83) - cylinder85) - cylinder86) - cylinder87) - cube161)) + ((((((((cylinder51 * linear_extrude57) * cube63) * rotate_extrude92) - cylinder86) - cylinder87) - cube161) + ((((((cylinder51 * linear_extrude57) * cylinder66) * rotate_extrude92) - cylinder86) - cylinder87) - cube161)) + ((((((cylinder51 * cylinder70) * cylinder73) * rotate_extrude92) - cylinder86) - cylinder87) - cube161))) + ((((((((((linear_extrude101 - cylinder136) - cube161) - cylinder341) - render820) - linear_extrude831) + (((cube106 - cylinder136) - cube161) - cylinder341)) + ((((cube108 - cylinder136) - cube161) - cylinder341) - linear_extrude831)) + ((((cylinder115 - cylinder121) - cylinder136) - cube161) + (((cylinder119 - cylinder121) - cylinder136) - cube161))) + (((cube125 - linear_extrude129) - cylinder136) - cube161)) + ((((((((((((linear_extrude101 - cube140) - linear_extrude144) - cube161) - cylinder341) - render820) - linear_extrude831) + ((cube106 - cube161) - cylinder341)) + (((cube108 - cube161) - cylinder341) - linear_extrude831)) + (((cylinder115 - cylinder121) - cube161) + ((cylinder119 - cylinder121) - cube161))) + ((((cube125 - linear_extrude129) - cube140) - linear_extrude144) - cube161)) + ((((((((((linear_extrude101 - cube140) - linear_extrude148) - cube161) - cylinder341) - render820) - linear_extrude831) + (((cube106 - linear_extrude148) - cube161) - cylinder341)) + ((((cube108 - linear_extrude148) - cube161) - cylinder341) - linear_extrude831)) + ((((cylinder115 - cylinder121) - linear_extrude148) - cube161) + (((cylinder119 - cylinder121) - linear_extrude148) - cube161))) + ((((cube125 - linear_extrude129) - cube140) - linear_extrude148) - cube161))) + (((((((((linear_extrude101 * linear_extrude155) - cube161) - cylinder341) - render820) - linear_extrude831) + (((cube106 * linear_extrude155) - cube161) - cylinder341)) + ((((cube108 * linear_extrude155) - cube161) - cylinder341) - linear_extrude831)) + ((((cylinder115 * linear_extrude155) - cylinder121) - cube161) + (((cylinder119 * linear_extrude155) - cylinder121) - cube161))) + (((cube125 * linear_extrude155) - linear_extrude129) - cube161))))) + (((((((((((((((cylinder173 - cylinder200) - cube218) - cube220) - cylinder223) - cube226) - cube228) - cylinder231) - cube234) - cube236) - cylinder239) - cylinder345) + (((((((cylinder179 - cylinder200) - cube218) - cylinder239) - cylinder345) + ((((cylinder182 - cylinder200) - cube220) - cylinder231) - cylinder345)) + ((((((cylinder187 - cylinder200) - cylinder223) - cube226) - cylinder239) - cylinder345) + (((cylinder190 - cylinder200) - cube228) - cylinder345))) + ((((cylinder195 - cylinder200) - cube234) - cylinder345) + (((((cylinder198 - cylinder200) - cylinder223) - cylinder231) - cube236) - cylinder345)))) + ((((cube204 - cylinder345) + (((cube207 - cylinder223) - cylinder231) - cylinder345)) + (((cube210 - cylinder223) - cylinder239) - cylinder345)) + (cube213 - cylinder345))) + (((((((((((((((((cylinder250 - cylinder285) - cube303) - cube305) - cylinder308) - cube311) - cube313) - cylinder316) - cube319) - cube321) - cylinder324) - cube327) - cube329) - cylinder332) - cylinder345) - render346) + (((((((((cylinder256 - cylinder285) - cube303) - cylinder324) - cube329) - cylinder345) + ((((((cylinder259 - cylinder285) - cube305) - cylinder316) - cube327) - cylinder345) - render346)) + ((((((cylinder264 - cylinder285) - cylinder308) - cube311) - cylinder324) - cylinder345) + (((((cylinder267 - cylinder285) - cube313) - cylinder332) - cylinder345) - render346))) + (((((cylinder272 - cylinder285) - cube319) - cylinder332) - cylinder345) + (((((cylinder275 - cylinder285) - cylinder308) - cylinder316) - cube321) - cylinder345))) + (((((((cylinder280 - cylinder285) - cube305) - cylinder316) - cube327) - cylinder345) - render346) + (((((cylinder283 - cylinder285) - cube303) - cylinder324) - cube329) - cylinder345)))) + (((((cube289 - cylinder345) - render346) + (((cube292 - cylinder308) - cylinder316) - cylinder345)) + (((cube295 - cylinder308) - cylinder324) - cylinder345)) + ((cube298 - cylinder332) - cylinder345)))) + (((cube337 - cylinder341) - cylinder345) - render346))) [... cut ...]  - cylinder758) - cylinder766) - cube771) - cylinder795))) + ((((((cylinder730 - cylinder735) - cube755) - cylinder766) - cube777) - cylinder795) + ((((((cylinder733 - cylinder735) - cube753) - cylinder774) - cube779) - cylinder795) - render796)))) + ((((cube739 - cylinder795) + (((cube742 - cylinder758) - cylinder766) - cylinder795)) + (((cube745 - cylinder758) - cylinder774) - cylinder795)) + ((cube748 - cylinder782) - cylinder795)))) + (((cube787 - cylinder791) - cylinder795) - render796)))))

MichaelAtOz, 2017-04-09:
This is that object, lots of fiddly bits...
[image]

MichaelAtOz, 2017-04-09:
Note my other [suggestion](http://forum.openscad.org/Console-information-td21111.html) to add timestamps to various console messages. This could at least make it easier to compare two different versions of code.

MichaelAtOz, 2017-04-09:
Note there is a bug in the example, it makes 4 fins, the 4th in the same spot as the 1st...

royasutton, 2017-04-09:
@MichaelAtOZ, thank you for the quite helpful example. It is not clear what the _multi-thread tree_ is (the CSG tree mapped to process threads?), but I will continue with my thoughts based on what you have exposed.

Knowing a breakdown of script resource usage to (1) generate the CSG Tree and (2a) preview (OpenCSG) or (2b) render (CGAL) would be good information to guide script optimization. It appears to me that what is needed is: **a mechanism to synchronize points in the CSG Tree with actual system resource usage.** 

Maybe the proposed resource usage query command could have an option that allows the invocation to force a render? The actual resources use would, of course, change based on OpenCSG vs CGAL. 

As for tracking the CPU-time consumed (_a resource metric_... and most likely of primary interest) by components of the multi-thread tree. My intuition tell me that OpenSCAD is not likely keeping track of the hierarchy of function/module invocation that would make the information more meaningful for _kernel_ optimization. What I am trying to identify is that a function or module may be called by many parts of a compiled script. But it is fundamentally important to know how much is attributed to some "_kernel_" of  interest that is in focus for optimization.

I am guessing that the cache matter is not that difficult. Add a command line option (with GUI equivalent) to disable the cache during a compile.

Are there any incremental controls to preview/render the CSG? For example, can one force that a preview/render at some arbitrary point within the CSG Tree? If so, then those controls could be used to gather the information. I am guessing the next issue would be getting the information back into the script since variable assignments were likely done at AST???

Hmmm...

Maybe the strategy could be to allow the resource query command to selectively annotate the CSG tree with resource usages (or even generate a separate file if need be). In any case, I think **the user needs control over what is being measured**. 

Should be able to assign measurement tags with measurement results (and preferably **any** of the standard resources the OS keeps track of)... As already pointed out, that information could then be post-processed to extract the metrics of interest.

royasutton, 2017-05-02:
Thanks to all for taking the time to consider this idea... It is clear to me that is much more involved than I had originally imagined!
Roy

closed 2017-05-02 by royasutton