openscad/openscad #1960: Feature request: script-level file output; ie: fecho("text");

royasutton, 2017-02-23:
New commands to _open_, _close_, and _write_ text information to a file.

Having only a single output stream (the console) greatly limits design flow options. This is why, for example, some have a separate BOM mode where design instance count are echoed to the console.

Having three new functions:
```
v = fopen(  "file-name"  );	
fclose( v );
fecho( v, "text" );
```
would open up many design flow options. 

A separate text stream could be opened for errors, statistics, counts, instructions, for example. The communication of design information become much more efficient. No longer are multiple passes required with separate console-output modes.

The OpenSCAD console support many HTML commands. Currently, I have to jump through many needless hoops to store script-created HTML tables to a file. It would be nice to be able to write HTML tables directly to a specified file by simple writing to the correct output stream.

royasutton, 2017-05-02:
Looks as though this is pretty far down on the interest list.
Roy

closed 2017-05-02 by royasutton

MichaelAtOz, 2017-05-02:
Roy,

You can get the console output to a file with the command line
"For at least 2015.03-2+, specifying the extension .echo causes openscad to produce a text file containing error messages and the output of all echo() calls in filename as they would appear in the console window visible in the GUI. Multiple output files are not supported, so using this option you cannot also obtain the model that would have normally been generated." (openscad somefile.scad -o console.echo)
If that helps...