openscad/openscad #1964: Windows cross compilation broken

buttim, 2017-02-28:
Following the reccomended procedure for cross compiling under Ubuntu I get the following error:

../src/LibraryInfo.cc:27:17: fatal error: zip.h: No such file or directory 

Modifying scripts/mingw-x-build-dependencies.sh and adding libzip to the end of lines 81 and 88, download-libzip to lines 79 and 87 the error disappears, only to be substituted by tons of errors the first of which is:

In file included from /usr/include/freetype2/ft2build.h:37:0,
                 from ../src/FreetypeRenderer.h:33,
                 from ../src/textnode.h:6,
                 from ../src/GeometryEvaluator.cc:17:
/usr/include/freetype2/freetype/config/ftheader.h:78:3: error: #error unknown multiarch location for ftheader.h

t-paul, 2017-02-28:
Please try using the `release-common.sh` as mentioned in the [forum post](http://forum.openscad.org/Windows-cross-compilation-tp20413p20418.html).

This is used to build the dev snapshots and works even with the still existing libzip issue.

buttim, 2017-02-28:
First I get the following messages

./scripts/release-common.sh: illegal option -- s
./scripts/release-common.sh: illegal option -- n
./scripts/release-common.sh: illegal option -- a
./scripts/release-common.sh: illegal option -- p
./scripts/release-common.sh: illegal option -- s
./scripts/release-common.sh: illegal option -- h
./scripts/release-common.sh: illegal option -- o
./scripts/release-common.sh: illegal option -- t

and finally 

make: *** No rule to make target 'release'.  Stop.

t-paul, 2017-02-28:
The initial warnings are normal, those are caused by the simple option parsing.

Please add some more of the actual output, it's not possible to help with just some final error message.

buttim, 2017-03-01:
didn't want to copy & paste a very long output.
Lots of rm -f for object files then

rm -f *~ core *.core
Building GUI binary...
make: *** No rule to make target 'release'.  Stop.
cant find release/openscad.exe. build failed. stopping.

donbright, 2017-04-12:
@buttim can you list ubuntu version you are using? thanks.

buttim, 2017-04-12:
Sure: 16.04

t-paul, 2018-04-06:
Closing as there's no more feedback and the build does work locally and on the build machine. Feel free to reopen to have another stab at the issue.

closed 2018-04-06 by t-paul