openscad/openscad #1965: Request: numerical font weight in text module

whosawhatsis, 2017-03-01:
It's easy for attempts to print text to fail because features are too thin. For some typefaces, this is unavoidable (I'm looking at you, Zapfino!), but it would be nice if there was a feature that, at least for the default font, would let you set the font weight (the thickness of the lines that make up the letters, assuming they are uniform in the chosen font), rather than just bold or regular, to a specific width that you know will be printable. This would take out a lot of the guesswork. It is possible to do this with offset, but it takes significant manual tuning to make it work.

MichaelAtOz, 2017-03-30:
+1. Significant PITA getting text past Shapeways thin wall detector...

jordanbrown0, 2024-09-22:
The documentation is not great, but OpenSCAD uses fontconfig to select fonts and so fontconfig selection specification are available.

`font=":weight=NNN"` will select a font with the specified weight, where NNN is a number from 0 to something like 215.

https://fontconfig.pages.freedesktop.org/fontconfig/fontconfig-user.html

closed 2024-09-22 by jordanbrown0