openscad/openscad #1966: Difference between cylinders

goganchic, 2017-03-04:
I want to draw difference between two cylinders. So, I use such code:

```
difference() {
    cylinder(h=100, r=100);
    cylinder(h=100, r=50);
}
```

I expect to see through hole, but the picture differs.

[image]

It looks like some pixels were not substracted from first cylinder. If I incease size of internal cylinder

```
difference() {
    cylinder(h=100, r=100);
    translate([0, 0, -1])
        cylinder(h=102, r=50);
}
```

then everything works fine

[image]

Is it ok or bug?

nophead, 2017-03-04:
It is a known limitation of OpenCSG. Coincident faces cause Z fighting because the Z values get rounded to integers in the hardware Z buffer of your graphics card.

goganchic, 2017-03-04:
Does some known workaround exists?

nophead, 2017-03-04:
Yes, exactly what you did, i.e. subtract a longer cylinder, so there are no coincident faces.

To avoid the translate I often centre the second cylinder and make its height twice the first one plus 1.

Or you can use render() to compute the geometry with CGAL, so the OpenCSG preview only has to draw a tube, not fake one by drawing two cylinders.

alexrj, 2017-03-04:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview.3F

goganchic, 2017-03-04:
ok, thanks!

closed 2017-03-04 by t-paul