openscad/openscad #1968: Strange behavior with special variables as module parameters

speendo, 2017-03-13:
I experienced a strange behavior when using special variables as module parameters. Looking at primitives like `sphere()` I thought this was a normal thing to do but maybe I am wrong...

However, here is a minimal example:

    $special = 4;

    module test($special = 2) {
        echo("test: ", $special);
    }

    special_value = $special;

    echo("global $special: ", $special);
    // ECHO: "global $special: ", 4 <= ok
    echo("global special_value: ", special_value);
    // ECHO: "global special_value: ", 4 <= ok

    test($special = special_value);
    // ECHO: "test: ", 4 <= ok

    test($special = $special);
    // ECHO: "test: ", 2 <= what?

I'm curious what you think about that!

nophead, 2017-03-14:
Looks like a bug to me.

MichaelAtOz, 2017-03-15:
```
$special = 4;

module test(module_scope_before=$special, $special = 2, module_scope_after=$special) {
    echo("test: ", $special, module_scope_before=module_scope_before,module_scope_after=module_scope_after);
}

special_value = $special;

echo("global $special: ", $special);
// ECHO: "global $special: ", 4 <= ok
echo("global special_value: ", special_value);
// ECHO: "global special_value: ", 4 <= ok

test($special = special_value);
// ECHO: "test: ", 4 <= ok
//ECHO: "test: ", 4, module_scope_before = 4, module_scope_after = 2

test($special = $special);
// ECHO: "test: ", 2 <= what?
//ECHO: "test: ", 2, module_scope_before = 4, module_scope_after = 2

test($special = 6);
//ECHO: "test: ", 6, module_scope_before = 4, module_scope_after = 2
```

I don't think it is meant to do that? module_scope_after should be 4 in all (I believe the parameters should use the scope at call time - i.e. the main scope in this case), if someone suggests it should be 2 in the 2nd, then it should be 6 in the 3rd.
Did the let() sequential evaluation escape?

kintel, 2017-03-18:
I suspect this to be introduced by #1923 - needs some looking into.

nophead, 2017-03-18:
I think it predates #1923 because 2016.10.04 (git 7e0935d) gets it wrong.

speendo, 2017-03-18:
I experienced this issue in OpenSCAD 2015.03 (Debian Stretch x64)

By the way: is there a real usecase for a module with a special variable as a parameter? Or is this something one shouldn't do anyway?

nophead, 2017-03-18:
I can't think of a use case, which is probably why nobody noticed the bug before. You can do this to avoid special parameters:

module test(special = 2) {
    $special = special;
    echo("test: ",$special);
}

MichaelAtOz, 2017-03-18:
I think the main use (tho as nophead shows there is a work-around) is with children()

speendo, 2017-03-18:
When I did it I tried to write some sort of wrapper module for cylinder (which uses $fa, $fs and $fn as parameters).

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#cylinder

But I guess there is another way to do so...

MichaelAtOz, 2017-03-18:
While others have disagreed on this method, you don't have to specify all parameters. Any extra parameters in the call become local variables to the module.
```
module myCylinder(r,h) {
  echo(r,h,$fn,yeti);
  cylinder(r=r,h=h,yeti);
}

myCylinder(5,10,yeti=true,$fn=64);
```

nophead, 2017-03-18:
Yes but that doesn't allow assigning a default for $fn, which is what causes the bug.

speendo, 2017-03-18:
@MichaelAtOz I see. Thank you for clearing this up for me. I thought there would be a method like this, however I also thought this was limited to special variables...

speendo, 2017-03-18:
@nophead I don't think so. See this example:

    $nodefault = 5;

    module test($nodefault) {
        echo($nodefault);
    }

    test($nodefault = $nodefault);

MichaelAtOz, 2017-03-18:
Let me post a tested version:

```
module myCylinder(r,h) {
  echo(r=r,h=h,$fn=$fn,yeti=yeti);
  cylinder(h=h,r=r,center=yeti);
}

myCylinder(5,10,yeti=true,$fn=64);
// ECHO: r = 5, h = 10, $fn = 64, yeti = true
```
Note without specifying $fn as a parameter in the module definition, $fn gets passed, as does yeti.
The bug is when you specify the $ variable as a parameter.

While it can be argued that non special variables using this is abusing it (I find it very useful), using $ variables is the way to feed input to children()

nophead, 2017-03-18:
It goes wrong even if you use $fn as a parameter, which is surely reasonable if you want to wrap cylinder and give $fn a default value.

The bug seems to be that when it looks up a special variable in the rhs of the parameter assignment it finds the parameter being assigned (which may have already been initialised to the default) instead of the global value of the same name. So the new parameter gets dynamic scope too early and affects is own initialisation.

thehans, 2017-06-04:
I was struggling with some special var issues in my own script recently, and finally narrowed down the cause.  

Here's the simplified example of what I was doing:
```
$fn = 100;
function getfn() = $fn; 

module busted(x, $fn=8) {
  echo("getfn() = ", x, "$fn = ", $fn);  
}

busted(getfn());
//ECHO: "getfn() = ", 8, "$fn=", 8
```

getfn is picking up busted's $fn value just from being called in the same parameter list?

nophead, 2017-06-05:
I think that is correct because $variables have dynamic scope. When getfn is called the inner most $fn is 8, so that is value it uses. Just the same as say circle().

thehans, 2017-06-05:
It doesn't seem correct to me, since getfn is not actually being called from busted's scope.

RonaldoCMP, 2017-06-05:
There is no correct or incorrect way here. It is a design choice. It may be
considered good or bad, though.

Note that in the definition of module busted() the setting $fn=8 is
internal to the module scope. So it is reasonable that the getfn() call be
internal too.

 Anyway, you always may evaluate getfn() before setting the busted()
parameter:

x = getfn();
busted(x);

MichaelAtOz, 2017-06-06:
It's a bug
```
$fn = 100;
function getfn() = $fn; 

module busted(x, $fn=8) {
  echo(str("getfn() = ", getfn() ), x = x, "$fn = ", $fn);  
}

busted(getfn());
//ECHO: "getfn() = 8", x = 8, "$fn=", 8
```
`getfn()` called within `busted()` should return the `$fn` from the scope of the function definition.

MichaelAtOz, 2017-06-06:
...and x should be 100 IMO.
But $variables behaviour (intent) has not been documented in detail.

RonaldoCMP, 2017-06-07:
Bug or not, this is exactly the behavior of special variables exemplified
in the wiki manual.

thehans, 2021-05-07:
Fixed by #3745

closed 2021-05-07 by thehans