openscad/openscad #1974: Projection seems inconsistent depending on projected content

christophevg, 2017-03-27:
Hi,

I've stumbled upon inconsistent behaviour of `projection(cut=true)`.

Given the following minimal model:

```
// doesn't render
projection(cut=true) {
  rotate([90,0,90]) {
                          cube([20, 5, 5]);
    translate([30, 0, 0]) cube([20, 5, 5]);
  }
}
```

This results in nothing being rendered.

> The example is the smallest version I could distill from a larger whole, so it might seem a bit strange.

Some observations:

1) When the rotation is removed, both cubes are rendered in the projection.

```
// renders
projection(cut=true) {
                        cube([20, 5, 5]);
  translate([30, 0, 0]) cube([20, 5, 5]);
}
```

But that is essential to the larger picture, so I can't easily leave that out.

2) Rewriting this example to implement the rotation directly also renders:

```
// renders
projection(cut=true) {
                          cube([5, 20, 5]);
    translate([0, 30, 0]) cube([5, 20, 5]);
}
```

Although it would be doable, I don't feel like rewriting the highly modular actual object that is here represented by the simple cube :-)

Especially not because in the end this doesn't seem to be the root cause, as illustrated by the third observation:

3) When removing/commenting out one of the two cubes, the projection renders:

```
// renders
projection(cut=true) {
  rotate([90,0,90]) {
                          cube([20, 5, 5]);
//  translate([30, 0, 0]) cube([20, 5, 5]);
  }
}
```

```
// renders
projection(cut=true) {
  rotate([90,0,90]) {
//                        cube([20, 5, 5]);
    translate([30, 0, 0]) cube([20, 5, 5]);
  }
}
```

It's because of the strange inconsistency between the two latter cases, that I decided to submit it as a bug. It seems that the projection can be made for each cube, but not for both at the same time ?!

I've discovered this problem when using the 2015 release, checked it with a 2018 more recent build and I've tried to compile the master branch, but failed to complete that.

Thanks for looking into this,
regards,
Christophe VG

MichaelAtOz, 2017-03-27:
Yeh, looks like a bug. Confirmed in 2015.03-2 & 2017.01.20 on W7/64.

Workaround:
```
// doesn't render
projection(cut=true) 
  {
  translate([0,0,-0.1])
    rotate([90,0,90]) 
    {
      cube([20, 5, 5]);
      translate([30, 0, 0]) 
        cube([20, 5, 5]);
    }
  }
```

MichaelAtOz, 2017-03-27:
Note it is both Preview & Render.

nophead, 2017-03-27:
When you use cut the projection is of the features on the plane at Z = 0. It isn't a good idea to have the cut plane coincident with a face of the object.

The reason it fails is because the built in rotate isn't accurate for 90 degrees. The original code works if you use a user space version like this:
```
module rotate(angle)            // built-in rotate is inaccurate for 90 degrees, etc
{
 a = len(angle) == undef ? [0, 0, angle] : angle;
 cx = cos(a[0]);
 cy = cos(a[1]);
 cz = cos(a[2]);
 sx = sin(a[0]);
 sy = sin(a[1]);
 sz = sin(a[2]);
 multmatrix([
  [ cy * cz, cz * sx * sy - cx * sz, cx * cz * sy + sx * sz, 0],
  [ cy * sz, cx * cz + sx * sy * sz,-cz * sx + cx * sy * sz, 0],
  [-sy,      cy * sx,                cx * cy,                0],
  [ 0,       0,                      0,                      1]
 ]) children();
}
```

That way the rotated cubes are exactly at z = 0, so the projection sees them.

nophead, 2017-03-28:
So actually there is nothing wrong with projection, rotate is the problem and actually all uses of trig functions in OpenSCAD that take degrees should use the accurate versions. At the moment only the expression evaluator uses them.

christophevg, 2017-03-28:
@nophead I had that feeling, but wasn't sure, because when only one cube was used, it works, for either one of them (observation 3). So it seemed to work separately, but not together. Which puzzled me. The rotation is(/should be?) the same, for one or two cubes.

t-paul, 2017-03-28:
Rotation issue is #517.

nophead, 2017-03-28:
@christophevg you would think the rotation would be the same but the object being rotated is very different. 

When you have two cubes the result is a CGAL union with vertices represented in CGALs numeric representation, probably infinite precision rationals. With only one cube CGAL is not involved and the result is represented with a different vertex format, probably doubles. I suspect in the later case the slightly off vertices are snapped back into position by rounding.

t-paul, 2020-02-16:
Closing as the original example renders now (due to some of the special rotations being handled separately) and there's another issue open to catch remaining rotation issues.

closed 2020-02-16 by t-paul