openscad/openscad #1975: A square changes behavior of a circle and $fn

hroncok, 2017-03-27:
On OpenSCAD 2015.03-3, Fedora 25 64bit.

Consider the following code:

```openscad
module fn(s=1) {
    scale(s) linear_extrude(2/s) {
        circle(r=10/s);
        //square(5/s, center=true);
    }
}

for (x=[0:7], y=[0:9]) {
    s = pow(10,x);
    f = 3*y+1;
    translate([40*x,40*y,0]) fn(s, $fn=f);
    echo("Coordinate", [x,y], "s", s, "$fn", f);
}
```

This is the result (perfectly reasonable):

[image]

Now I uncomment the `square` and this is the result:

[image]

Notice that the `$fn` starts acting weird.  In the forth column, if you red the picture form bottom, the resolution is higher and higher until it goes down and down. Not to mention the missing columns :(

Logged echos in both cases:

<details>

```
ECHO: "Coordinate", [0, 0], "s", 1, "$fn", 1
ECHO: "Coordinate", [0, 1], "s", 1, "$fn", 4
ECHO: "Coordinate", [0, 2], "s", 1, "$fn", 7
ECHO: "Coordinate", [0, 3], "s", 1, "$fn", 10
ECHO: "Coordinate", [0, 4], "s", 1, "$fn", 13
ECHO: "Coordinate", [0, 5], "s", 1, "$fn", 16
ECHO: "Coordinate", [0, 6], "s", 1, "$fn", 19
ECHO: "Coordinate", [0, 7], "s", 1, "$fn", 22
ECHO: "Coordinate", [0, 8], "s", 1, "$fn", 25
ECHO: "Coordinate", [0, 9], "s", 1, "$fn", 28
ECHO: "Coordinate", [1, 0], "s", 10, "$fn", 1
ECHO: "Coordinate", [1, 1], "s", 10, "$fn", 4
ECHO: "Coordinate", [1, 2], "s", 10, "$fn", 7
ECHO: "Coordinate", [1, 3], "s", 10, "$fn", 10
ECHO: "Coordinate", [1, 4], "s", 10, "$fn", 13
ECHO: "Coordinate", [1, 5], "s", 10, "$fn", 16
ECHO: "Coordinate", [1, 6], "s", 10, "$fn", 19
ECHO: "Coordinate", [1, 7], "s", 10, "$fn", 22
ECHO: "Coordinate", [1, 8], "s", 10, "$fn", 25
ECHO: "Coordinate", [1, 9], "s", 10, "$fn", 28
ECHO: "Coordinate", [2, 0], "s", 100, "$fn", 1
ECHO: "Coordinate", [2, 1], "s", 100, "$fn", 4
ECHO: "Coordinate", [2, 2], "s", 100, "$fn", 7
ECHO: "Coordinate", [2, 3], "s", 100, "$fn", 10
ECHO: "Coordinate", [2, 4], "s", 100, "$fn", 13
ECHO: "Coordinate", [2, 5], "s", 100, "$fn", 16
ECHO: "Coordinate", [2, 6], "s", 100, "$fn", 19
ECHO: "Coordinate", [2, 7], "s", 100, "$fn", 22
ECHO: "Coordinate", [2, 8], "s", 100, "$fn", 25
ECHO: "Coordinate", [2, 9], "s", 100, "$fn", 28
ECHO: "Coordinate", [3, 0], "s", 1000, "$fn", 1
ECHO: "Coordinate", [3, 1], "s", 1000, "$fn", 4
ECHO: "Coordinate", [3, 2], "s", 1000, "$fn", 7
ECHO: "Coordinate", [3, 3], "s", 1000, "$fn", 10
ECHO: "Coordinate", [3, 4], "s", 1000, "$fn", 13
ECHO: "Coordinate", [3, 5], "s", 1000, "$fn", 16
ECHO: "Coordinate", [3, 6], "s", 1000, "$fn", 19
ECHO: "Coordinate", [3, 7], "s", 1000, "$fn", 22
ECHO: "Coordinate", [3, 8], "s", 1000, "$fn", 25
ECHO: "Coordinate", [3, 9], "s", 1000, "$fn", 28
ECHO: "Coordinate", [4, 0], "s", 10000, "$fn", 1
ECHO: "Coordinate", [4, 1], "s", 10000, "$fn", 4
ECHO: "Coordinate", [4, 2], "s", 10000, "$fn", 7
ECHO: "Coordinate", [4, 3], "s", 10000, "$fn", 10
ECHO: "Coordinate", [4, 4], "s", 10000, "$fn", 13
ECHO: "Coordinate", [4, 5], "s", 10000, "$fn", 16
ECHO: "Coordinate", [4, 6], "s", 10000, "$fn", 19
ECHO: "Coordinate", [4, 7], "s", 10000, "$fn", 22
ECHO: "Coordinate", [4, 8], "s", 10000, "$fn", 25
ECHO: "Coordinate", [4, 9], "s", 10000, "$fn", 28
ECHO: "Coordinate", [5, 0], "s", 100000, "$fn", 1
ECHO: "Coordinate", [5, 1], "s", 100000, "$fn", 4
ECHO: "Coordinate", [5, 2], "s", 100000, "$fn", 7
ECHO: "Coordinate", [5, 3], "s", 100000, "$fn", 10
ECHO: "Coordinate", [5, 4], "s", 100000, "$fn", 13
ECHO: "Coordinate", [5, 5], "s", 100000, "$fn", 16
ECHO: "Coordinate", [5, 6], "s", 100000, "$fn", 19
ECHO: "Coordinate", [5, 7], "s", 100000, "$fn", 22
ECHO: "Coordinate", [5, 8], "s", 100000, "$fn", 25
ECHO: "Coordinate", [5, 9], "s", 100000, "$fn", 28
ECHO: "Coordinate", [6, 0], "s", 1e+06, "$fn", 1
ECHO: "Coordinate", [6, 1], "s", 1e+06, "$fn", 4
ECHO: "Coordinate", [6, 2], "s", 1e+06, "$fn", 7
ECHO: "Coordinate", [6, 3], "s", 1e+06, "$fn", 10
ECHO: "Coordinate", [6, 4], "s", 1e+06, "$fn", 13
ECHO: "Coordinate", [6, 5], "s", 1e+06, "$fn", 16
ECHO: "Coordinate", [6, 6], "s", 1e+06, "$fn", 19
ECHO: "Coordinate", [6, 7], "s", 1e+06, "$fn", 22
ECHO: "Coordinate", [6, 8], "s", 1e+06, "$fn", 25
ECHO: "Coordinate", [6, 9], "s", 1e+06, "$fn", 28
ECHO: "Coordinate", [7, 0], "s", 1e+07, "$fn", 1
ECHO: "Coordinate", [7, 1], "s", 1e+07, "$fn", 4
ECHO: "Coordinate", [7, 2], "s", 1e+07, "$fn", 7
ECHO: "Coordinate", [7, 3], "s", 1e+07, "$fn", 10
ECHO: "Coordinate", [7, 4], "s", 1e+07, "$fn", 13
ECHO: "Coordinate", [7, 5], "s", 1e+07, "$fn", 16
ECHO: "Coordinate", [7, 6], "s", 1e+07, "$fn", 19
ECHO: "Coordinate", [7, 7], "s", 1e+07, "$fn", 22
ECHO: "Coordinate", [7, 8], "s", 1e+07, "$fn", 25
ECHO: "Coordinate", [7, 9], "s", 1e+07, "$fn", 28
```

</details>

t-paul, 2017-03-27:
#999 maybe?

hroncok, 2017-03-27:
Sure. And I was so happy about my nice reproducer :D

closed 2017-03-27 by hroncok

t-paul, 2017-03-27:
Sorry, nice report. It's now referenced from the other issue too, so it can still serve as test case for the test suite when fixing that.