openscad/openscad #1977: color is being applied inconsistently

tcurdt, 2017-03-28:
All the cubes should have the top red and the bottom half blue.

```
module top1()
{
  translate([0,0,+10/2]) cube([20,20,10], center=true);
}

module bottom1()
{
  translate([0,0,-10/2]) cube([20,20,10], center=true);
}

module container()
{
  cube([20,20,20], center=true);
}

module top2()
{
  difference() {
    container();
    translate([0,0,-15]) cube([30,30,30], center=true);
  }
}
module bottom2()
{
  difference() {
    container();
    translate([0,0,+15]) cube([30,30,30], center=true);
  }
}

module test11()
{
  color("red") top1();
  color("blue") bottom1();
}

module test12()
{
  color("red") top1();
  color("blue") bottom2();
}

module test22()
{
  color("red") top2();
  color("blue") bottom2();
}

module test21()
{
  color("red") top2();
  color("blue") bottom1();
}

d = 25;
translate([+d,0,+d]) test11();
translate([+d,0,-d]) test12();
translate([-d,0,+d]) test21();
translate([-d,0,-d]) test22();

translate([-2*d,0,+d]) top1();
translate([+2*d,0,+d]) top2();
translate([-2*d,0,-d]) bottom1();
translate([+2*d,0,-d]) bottom2();
```

tcurdt, 2017-03-28:
[image]

MichaelAtOz, 2017-03-28:
[Z-fighting](https://en.wikipedia.org/wiki/Z-fighting)
Whenever you see this:
[image]
Because in preview with CSG, both objects are there, but it doesn't know which to show when they have co-incident faces.

There may be internal ways to fix it but whenever I get color issues I `render()` inside `color()`

Try
```
module bottom2()
{
render()
  difference() {
    container();
    translate([0,0,+15]) cube([30,30,30], center=true);
  }
}
```
Same for top2
[image]

t-paul, 2017-03-28:
I don't think there's a automatic way to resolve this. I've seen this in expensive CAD packages too which would have had the resources to fix it if possible. I'm closing this for now, in case there are any pointers to strategies how to improve this behavior, feel free to comment and reopen.

closed 2017-03-28 by t-paul

tcurdt, 2017-03-28:
I don't see `render` as a fix. I am also not sure this really is z-fighting (the way I know it) because there are no coplanar polygons here. Let me adjust the testcase a little to show:

```
d = 25;

module top1()
{
  translate([0,0,+10/2]) cube([20,20,10], center=true);
}

module bottom1()
{
  translate([0,0,-10/2]) cube([20,20,10], center=true);
}

module container()
{
  cube([20,20,20], center=true);
}

module top2()
{
  difference() {
    container();
    translate([0,0,-15]) cube([30,30,30], center=true);
  }
}
module bottom2()
{
  difference() {
    container();
    translate([0,0,+15]) cube([30,30,30], center=true);
  }
}

module test11()
{
  color("red") top1();
  color("blue") translate([0,0,-d]) bottom1();
}

module test12()
{
  color("red") top1();
  color("blue") translate([0,0,-d]) bottom2();
}

module test22()
{
  color("red") top2();
  color("blue") translate([0,0,-d]) bottom2();
}

module test21()
{
  color("red") top2();
  color("blue") translate([0,0,-d]) bottom1();
}

x = 40;
translate([0*x,0,0]) test11();
translate([1*x,0,0]) test12();
translate([2*x,0,0]) test21();
translate([3*x,0,0]) test22();
```

This looks like it should be:
[image]

tcurdt, 2017-03-28:
Here I visualized the blocks that cut the object in half. 
```
module top2()
{
  difference() {
    container();
    %translate([0,0,-15]) cube([30,30,30], center=true);
  }
}
module bottom2()
{
  difference() {
    container();
    %translate([0,0,+15]) cube([30,30,30], center=true);
  }
}
```

(z-fighting on the object should not be relevant here)

[image]

tcurdt, 2017-03-28:
Now I change it to `d = 5;` and it looks like this:

[image]

Which to me doesn't look like z-fighting to me. The cube has already been cut off before - but somehow it still affect the results.

tcurdt, 2017-03-28:
I can see how rendering would be confused when there are coplanar polygons - but how is this also a problem with the hidden/cutoff parts?

nophead, 2017-03-28:
>The cube has already been cut off before - but somehow it still affects the results.

Nothing is actually cut of with OpenCSG preview. It draws all the original cubes in the right order and graphics modes to create the end result. So you can get Z fighting between faces that don't exists in the final view.

I always render before colouring to get nice looking objects that draw quickly. If you have scenes like mine, which are machines with hundreds of parts, I have to render them or the preview gets too slow to draw.

tcurdt, 2017-03-28:
I see. That is pretty crazy.

@nophead you use `render` to speed things up? I was expecting the exact opposite.

When the subtree of the `render` childs is big and complex - doesn't that mean everything gets re-rendered on every preview change? Which for some of my models takes quite long and is only bearable on the final export. I would expect the same is true for you then.

t-paul, 2017-03-28:
Yes, the whole point of preview is to not actually calculate the result mesh - see http://opencsg.org/ which also links to some of the theory.

So render in general is usually slower, but it can speed up things if the rendered result reduces the amount of vertices or number of redraws (due to difference / intersection) as the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_the_preview_so_slow.3F) tries to explain in a slightly awkward way.

tcurdt, 2017-03-28:
> Yes, the whole point of preview is to not actually calculate the result mesh - see http://opencsg.org/ which also links to some of the theory.

Understood.

@t-paul So `render` will help me to not start crying when trying to rotate the preview ;) But it more or less means to render the subtree on each preview change - which in turn means waiting for the preview. Which I am not quite sure is better from my personal POV (or the crying).

tcurdt, 2017-03-28:
I also just tried to use `render` on a subtree but that result is even less expected behaviours (to me). Suddenly parts become "invisible" that should not.

tcurdt, 2017-03-28:
Yeah, the caching seem to work much better than expected. It seems like using `difference` after `render` is a problem though. Something along the lines of:
```
  difference() {
    union() {
      color("red") render() top();
      color("blue") render() bottom();
    }
    translate([0,-100/2,-100/2]) cube([100,100,100]);
  }
```
Which works great without the `difference`.
But with the cross-section I can see into the back of the object.

It should like this:

[image]

but looks like this:

[image]

nophead, 2017-03-28:
Yes, as I mentioned above, you need to specify convexity in render if you are going to do further CSG operations on it. The same applies to linear_extrude, etc. I.e. anything that isn't directly handled by OpenCSG.

tcurdt, 2017-03-28:
@nophead aaaaaah - on `render` itself! I thought on `linear_extrude` etc.

nophead, 2017-03-28:
Yes, basically anything that presents an arbitrary polyhedron to OpenCSG needs to specify its convexity. 

Shapes built from primitives combined with CSG ops don't.

tcurdt, 2017-03-28:
Makes sense.