openscad/openscad #1978: colors when using difference

tcurdt, 2017-03-28:
I am cutting an object in half which then removes the color on the cut surface.

[image]

MichaelAtOz, 2017-03-28:
OpenSCAD shows a difference()'d surface in another color if not specified.

See this example (also shows how to get rid of z-fighting)
```
difference() {
  color("blue") cube(10);
  translate([5,0,0])
    color("red") cube(10);
}

translate([0,20,0])
  difference() {
    color("blue") cube(10);
    translate([5,-1,-1])
      color("red") cube(12);
  } 

translate([0,40,0])
  difference() {
    color("blue") cube(10);
    translate([5,-1,-1])
      color("blue") cube(12);
  } 
```
[image]

thehans, 2017-03-28:
Also see https://github.com/openscad/openscad/issues/501

t-paul, 2017-03-28:
Closing as duplicate.

closed 2017-03-28 by t-paul