openscad/openscad #1982: Draft/fine rendering button missing

thomaskilian, 2017-03-31:
Currently there's just a single Render button. When designing complex models you definitely want to reduce rendering time by setting $fn to a low value, but something ready to export should have a high $fn value. Currently I un-/comment a single line, but often I forget to toggle that. So I either sit and wait too long for rendering or even worse end up with a draft rendering being exported for a print. So having two buttons with individual $fn settings would really be nice.

nophead, 2017-03-31:
This is a solution if you can use F5 for the low res preview: https://github.com/openscad/openscad/pull/1963

thomaskilian, 2017-03-31:
Hmm. I don't see any difference when using F5/F6 except that F5 shows "Preview" eq. to F9  and F6 "Solid Planes" eq. to F10 (I'm using German localization, so guess that's the translation). The Preview itself is not really useful compared to Solid since it has a wrong/bad rendering.

nophead, 2017-03-31:
I have never understood exactly what F9 and F10 do. I think they are just the antidotes to F11 and F12.

F5 and F6 should look identical except F5 can have colours and F6 is always one colour. F5 can have Z fighting due to coincident faces but I avoid those or use render() to sort them out. I don't know if that is what you mean by "wrong/bad rendering"?

This is the part I am working on at the moment and I can't see a single pixel change when I toggle between F5 and F6 but with $preview I could make F6 higher resolution if I needed to. I normally print with the same resolution as I use with F5. For example the faceting on this part is not visible when printed with ABS.

[image]

thomaskilian, 2017-03-31:
F5: [image]
vs
F6: [image]
But what I'm after is an automatic adjustment of $fn (here with 20):
[image]
The latter takes much longer for rendering (you might only notice the differences in the small bore holes).

nophead, 2017-03-31:
Your F5 view only looks wrong because you have some coincident faces. For example if you make a hole by subtracting a cylinder you need to make the cylinder longer. Or if you have a union of a cone on top of a cylinder you need to overlap them slightly or use hull instead of union.

Not sure why your part takes a long time to render with $fn set to only 20. My example has $fa = 6 and $fs = 0.25 and only takes 3 seconds.

t-paul, 2019-11-21:
Existing features (`$preview`) and following suggestions from comments above and [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?) should cover all the cases.

closed 2019-11-21 by t-paul