openscad/openscad #1983: Feature request: Use a better mesh for spheres

bramcohen, 2017-04-02:
The current method of tesselating spheres isn't very good. There are several better ones, as explained here:

https://gamedevdaily.io/four-ways-to-create-a-mesh-for-a-sphere-d7956b825db4

I'd recommend spherified cube, but really anything would be better than what it's using now.

thehans, 2017-04-02:
Its possible to implement in userspace.  The psuedocode from the article isn't the greatest example, but I managed to get a working version of the "normalized cube".  He does not explain the math behind any of it, and neglects to define functions like get_origin, get_right, and get_up.  I found that I had to subtract half the div_count from the iterators i and j, to center the shape for normalized cube to appear correct.

Here is the code:
https://gist.github.com/thehans/052129033443d8a8c14ac1e69b42e8da

I also attempted the "spherified cube", but could not make sense of the pseudocode.  I don't know what ```p2 = p * p``` is supposed to mean.  Cross product would result in zero vector, and dot product would be a scalar, so i guess its supposed to be pointwise product?   I'm not sure if the same halfway offset is needed on this one, but removing it looks a little more interesting though both are wildly incorrect.

From left to right: builtin sphere module, normalized_cube, spherified_cube
[image]

thehans, 2017-04-02:
OK, i got it working.  I found this article that had well written equations to go off of: http://catlikecoding.com/unity/tutorials/cube-sphere/ 
[image]

Code is in latest revision of the gist
https://gist.github.com/thehans/052129033443d8a8c14ac1e69b42e8da

MichaelAtOz, 2017-04-02:
So why is having tectonic fault lines better?

thehans, 2017-04-03:
I played around some more today and got icosahedrons to work too.
[image]

gist updated: https://gist.github.com/thehans/052129033443d8a8c14ac1e69b42e8da

MichaelAtOz, 2017-04-03:
Icosahedron looks good.

Performance 
```
type	r/fn-dc	facets	F5 sec	F6 sec	Vert	H/Edges	Edges	H/Facet	Facets
s	9/80	6242
nc	9/20	4800
sc	9/20	4800
ih	9/4	5120

s	90/160	25282	1	1
nc	90/40	19200	3	3
sc	90/20	19200	3	3
ih	90/5	20480	4	4

Intersection with cube(100)
s	90/160		1	42	1683	9690	4845	6328	3164
nc	90/40		3	28	1262	7326	3663	4806	2403
sc	90/40		3	29	1262	7326	3663	4806	2403
ih	90/5		4	38	1378	7974	3987	5222	2611
```

bobm123, 2017-04-12:
Would it be possible to adjust the position of the vertices that lie on the XY, ZX and YZ planes so they are evenly space around those circles? Doing so would let spheres line up with the cylinders used for making a rounded box as in MCAD boxes.scad (https://github.com/openscad/MCAD/blob/master/boxes.scad) 

A simpler example would be this:

```
$fn=16;
cylinder(r=9, h=15);
translate([0,0,15]) difference () {
    poly3d(spherified_cube(9,div_count=$fn/4));
    translate([0,0,-15]) cube(30, center=true);
}
```

[image]

doug-moen, 2017-04-12:
> MCAD/blob/master/boxes.scad)

"Would it be possible to adjust the position of the vertices that lie on
the XY, ZX and YZ planes so they are evenly space around those circles?"

Sounds like you really want a spherified octahedron, not a spherified cube.

doug-moen, 2017-04-12:
"Would it be possible to adjust the position of the vertices that lie on
the XY, ZX and YZ planes so they are evenly space around those circles?"

To clarify, if you adjust the vertices like this, then you will necessarily
have vertices inline with all of the axes at +X, -X, +Y, -Y, +Z, -Z, which
are the vertices of an octahedron. The number of vertices in each of the
great circles around the XY, ZX and YZ planes will necessarily be multiples
of 4. You'll end up with a spherified octahedron. The vertices will line up
with the sides of a cylinder if $fn is a multiple of 4.

bobm123, 2017-04-12:
Good point @doug-moen. Taking the original shperified_cube() it would have to use a different loops subdivide the face_points and of course the rotations for points and indexing for faces would change, but that will give me something to think about on the drive into work.

MichaelAtOz, 2017-04-13:
> spheres line up with the cylinders

Or if you want to be lazy, just hull() them...

bramcohen, 2017-04-13:
Using the technique for icosahedrons on octahedra should work well. It will be a bit less round than icosahedra but maybe a bit prettier and more intuitive. It might also be possible to allow for more fine grained control of the number of triangles. That might be true of icosahedra as well.

bobm123, 2017-04-13:
Hull will work of course if I just needed a rounded box, but I think an underlying idea of the original post was to avoid using CGAL. Those examples were just a way of exploring some of the possible use cases for these alternate sphere definitions.

It seems a little tricky to subdivide the triangular faces found of icosahedrons and octahedra. Cubes fit nicely into a nested for-loop. For that reason I was thinking a rhombic dodecahedron might be an interesting approach. Looking at the code for spherified_cube(), it seems like it could be modified for subdividing rhomboic faces and extended to generate the 12 rotations for points and then to generate the lists for faces. I'll post if I can get that working.

bobm123, 2017-04-14:
I had some success using a rhombic dodecahedron. I wasn't able to get the spherified version working, but normalized looks pretty good to me. I'm not sure if it performs any better than other polyhedra but it was an interesting way to explore the nice code written by @thehans. 

It turned out to be just a matter of replacing the unit vectors right=[1,0,0] and up=[0,1,0] with [1/2,-sqrt(2)/4,0] and [1/2,sqrt(2)/4,0], then replacing the rotations for cube faces with 12 new ones. The faces generator hardly changes at all, I just had to make it iterate to 12 instead of 6. 

In case you're interested, the code is here along with my cylinder examples from a few days ago:
https://gist.github.com/bobm123/d09ad2af5474cc5a28ef1a0fed161b12

[image]
From left to right, library sphere(), a rhombic dodecahedron, normalized RD with div_count=1 and normalized RD with div_count=10

mrvn, 2018-07-09:
Would it be possible to choose the start shape for division as argument to sphere()? Default could go from pyramid to icosahedron as $fn grows and then sub divide.

richfelker, 2020-01-21:
The original link is dead and wayback machine seems to have no archive of it. :-(

bramcohen, 2020-01-21:
The link in the top post of this thread is (was) nowhere near as good as the code provided in the responses, so you can just go with those.

richfelker, 2020-01-21:
Fair enough, but I was still interested in reading the writeup with it.

MichaelAtOz, 2020-01-21:
@richfelker Google is your friend, I think this is likely the one
https://medium.com/game-dev-daily/four-ways-to-create-a-mesh-for-a-sphere-d7956b825db4

thehans, 2020-01-21:
Also see the link from my comment where I found equations that were more correct than the original link: https://github.com/openscad/openscad/issues/1983#issuecomment-290975605

BTW, I guess I hadn't mentioned it before, but I ended up adding the alternative_spheres code as part of my 
"[FunctionalOpenSCAD](https://github.com/thehans/FunctionalOpenSCAD)" library which others might find useful.

Sarah-C, 2022-03-06:
Just passing through to say - Thank you everyone for this work. It may not be in the release, but it's still very important/useful for users. It's appreciated!

MarkJeronimus, 2024-02-03:
> Would it be possible to adjust the position of the vertices that lie on the XY, ZX and YZ planes so they are evenly space around those circles? Doing so would let spheres line up with the cylinders used for making a rounded box as in MCAD boxes.scad

I stumbled upon this too in the past, and only recently I acquired the skills to pull it off:
```scad
$fn = 24;

isosphere(d=200);

module isosphere(r=undef, d=undef) {
    radius    = r?r:d/2;
    divisions = ceil($fn/4);

    vertices = [[ 0,  0, -1],
                [ 0, -1,  0],
                [ 1,  0,  0],
                [ 0,  1,  0],
                [-1,  0,  0],
                [ 0,  0,  1]];
    faces = [[0, 1, 2],
             [0, 2, 3],
             [0, 3, 4],
             [0, 4, 1],
             [2, 1, 5],
             [3, 2, 5],
             [4, 3, 5],
             [1, 4, 5]];

    // For each triangular face, choose two sides as basis vectors (parameter 'order'), then convert
    // triangle coordinates [x, y] (where y<=x) to distances along the basis vectors (parameter 'mat')
    // in the specific order that the vertices should be.
    // This needs to be done three times, rotationally symmetric around 120 deg, because just redistributing
    // u and v with one set of basis vecors does not yet create a uniform distribution. Averaging these three
    // symmetries is somehow uniform, and there's no need to calculate the three mirror variants.
    // I did test this and normal and mirror variants were equal.
    sub_vertices_a = subdivide_faces(vertices, faces, divisions, [0, 1, 2], [[ 0,  1, 0], [ 1,  0, 0]]);
    sub_vertices_b = subdivide_faces(vertices, faces, divisions, [1, 2, 0], [[ 1, -1, 1], [ 0, -1, 1]]);
    sub_vertices_c = subdivide_faces(vertices, faces, divisions, [2, 0, 1], [[-1,  0, 1], [-1,  1, 0]]);
    all_vertices = [
        for (i = [0:len(sub_vertices_a)])
            normalize(sub_vertices_a[i]+sub_vertices_b[i]+sub_vertices_c[i])*radius
    ];
//    sub_vertices_d = subdivide_faces(vertices, faces, divisions, [0, 2, 1], [[0, 1, 0], [-1, 1, 0]]);
//    sub_vertices_e = subdivide_faces(vertices, faces, divisions, [1, 0, 2], [[1, -1, 1], [1, 0, 0]]);
//    sub_vertices_f = subdivide_faces(vertices, faces, divisions, [2, 1, 0], [[-1,  0, 1], [ 0, -1, 1]]);
//    all_vertices_mirror = [
//        for (i = [0:len(sub_vertices_a)])
//            normalize(sub_vertices_d[i]+sub_vertices_e[i]+sub_vertices_f[i])*r
//    ];

    face_vertex_indices_bottom = bottom_vertices(divisions);
    face_vertex_indices_top = top_vertices(divisions);
    face_vertex_indices = concat(face_vertex_indices_bottom, face_vertex_indices_top);

    sub_faces = make_faces(face_vertex_indices, divisions);

    polyhedron(all_vertices, sub_faces);
}

function subdivide_faces(vertices, faces, divisions, order, mat) = [
    vertices[0],
    for (i = [0:7])
        let (face = faces[i],
             p0 = vertices[face[order[0]]],
             pu = vertices[face[order[1]]]-vertices[face[order[0]]],
             pv = vertices[face[order[2]]]-vertices[face[order[1]]])
        for (y = [1:divisions-(i<4?0:1)],
             x = [1:y])
            let(uv = mat*[x, y, divisions])
            p0 + pu*redistribute(uv[0]/divisions) + pv*redistribute(uv[1]/divisions),
    vertices[5]
];

function bottom_vertices(divisions) = 
    let (tri_bot = divisions*(divisions+1)/2) [
    for (i = [0:3]) flatten([
        0,                             // Common bottom vertex,
        for (y = [1:divisions]) [      // For each row:
            (i+3)%4*tri_bot+y*(y+1)/2, // - Vertex from left-bordering face
            for (x = [1:y])            // - Vertices in this face
                tri_bot*i+(y-1)*y/2+x
        ]
    ])
];

function top_vertices(divisions) = 
    let (tri_top = (divisions-1)*divisions/2,
         tri_bot = divisions*(divisions+1)/2) [
    for (i = [4:7]) flatten([
        tri_bot*(i-3),                 // Bottom-right vertex from bottom-bordering face
        for (y = [1:divisions-1]) [    // For each row:
            tri_bot*(i-3)-y,           // - Vertex from bottom-bordering face,
            for (x = [1:y])            // - Vertices in this face
                tri_bot*4+tri_top*(i-4)+y*(y-1)/2+x
        ],
        ((i+3)%4+1)*tri_bot,           // Vertex from lower-left bordering face
        for (x = [1:divisions-1])      // Vertices from the left-bordering face
           tri_bot*4+(i+3)%4*tri_top+x*(x+1)/2,
        (tri_bot+tri_top)*4+1
    ])
];

function make_faces(face_vertex_indices, divisions) = flatten([
    for (vertices = face_vertex_indices)
        for (y = [1:divisions]) [
            for (x = [1:y])
                let(v1 = (y-1)*y/2+x-1,
                    v2 = y*(y+1)/2+x-1,
                    v3 = y*(y+1)/2+x)
                [vertices[v1], vertices[v2], vertices[v3]],
            if (y < divisions)
                for (x = [1:y])
                    let(v1 = y*(y+1)/2+x-1,
                        v2 = (y+1)*(y+2)/2+x,
                        v3 = y*(y+1)/2+x)
                    [vertices[v1], vertices[v2], vertices[v3]]
        ]
]);

function redistribute(x) = (tan(x*90-45)+1)/2;
function normalize(v) = v/norm(v); // convert vector to unit vector
function flatten(l) = [ for (a = l) for (b = a) b ];
```

Some development screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

lgarron, 2024-10-14:
I'd be really glad to see a better mesh for spheres, either by default or through a parameter!

I've recently been doing some geometry with spheres, and I've had to crank up the `$fn` to get enough fidelity near the poles.

I just tried out @MarkJeronimus's `isosphere(…)` and the code generates a few warnings for me but it has a nice result! 🤩

Although I do think this one is the most mathematically and aesthetically pleasing:
<img alt="geosphere11" width="960" src="https://private-user-images.githubusercontent.com/5273753/302063445-1e64ce5f-93e9-4709-bed7-3c5695587fbb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5MzA0NzAsIm5iZiI6MTcyODkzMDE3MCwicGF0aCI6Ii81MjczNzUzLzMwMjA2MzQ0NS0xZTY0Y2U1Zi05M2U5LTQ3MDktYmVkNy0zYzU2OTU1ODdmYmIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMTRUMTgyMjUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTgyN2NmOTIzM2I1NTZlNWRjYmQ5YjFlMDA4NjA4M2NkOTA0MjQzYjVmNTk1ZWE5ODMzODAzYzExMjI4MDZmZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.Us18nYHyTEAmm8o6t9_TnZnbrcFpVuDAHk7F9GWuZlQ">

MarkJeronimus, 2024-10-14:
@lgarron What warnings are you getting? I'm getting none.

Your private-user-image also doesn't work.

nophead, 2024-10-14:
I just rotate extrude a semicircle that has fn divisible by 4. I define
that as a sphere that overrides the built in version.

lgarron, 2024-10-14:
> I just rotate extrude a semicircle that has fn divisible by 4. I define that as a sphere that overrides the built in version.

Yeah, this is in fact what I ended up doing. 🙈

I can certainly work around this, but it would certainly be nice if others don't get caught out for literal edge cases like this that don't have a conceptual geometric issue.

lgarron, 2024-10-15:
> @lgarron What warnings are you getting? I'm getting none.

[image]

> Your private-user-image also doesn't work.

Hmm, not sure what broke. It's the second-to-last image from your post!

MarkJeronimus, 2024-10-15:
@lgarron Your line numbers are different from mine so I still can't help you troubleshoot. But if it's only warnings as you said in your first comment and it works out for you, then I'm happy to have helped. (If there's a blocking error like in your second post, then please retry with my exact code in a clean sketch)

&lt;Random&gt; Latest project I'm in the middle of, that uses it at a three-cylinder intersection (as that's what it's intended for after all):
[image]

MarkJeronimus, 2025-08-16:
For those still reading this and using my "isosphere" code, I found a bug, which is possibly the bug Igarron found, but possibly not as in my case I got errors and no geometry.

`for (i = [0:len(sub_vertices_a)])`

should be 

`for (i = [0:len(sub_vertices_a)-1])`

I have no idea why it worked for some inputs.

lgarron, 2026-08-25:
> There is a decent argument that anything that can be done in userspace should be done in userspace, but absent a solid bundled library that means that beginners are consigned to crummy meshes.

I think it would be nice to have an official decision either way, rather than the issue being closed by a bot for being stale.

This is certainly a feature I'd find really valuable to have in the core language. It is very rare that I code a sphere that I wouldn't want to use better geometry for.

MichaelAtOz, 2026-08-25:
> This is certainly a feature I'd find really valuable to have in the core language. It is very rare that I code a sphere that I wouldn't want to use better geometry for.

> but absent a solid bundled library that means that beginners are consigned to crummy meshes.

The BOSL2 [solid/unbundled] library has a few styles of [spherical object with controllable triangulation](https://github.com/BelfrySCAD/BOSL2/wiki/shapes3d.scad#functionmodule-sphere)
"triangulation of this spheroid can be controlled via the style= argument, where the value can be one of "orig", "aligned", "stagger", "octa", or "icosa""

[image]

octa, stagger, icosa.

jordanbrown0, 2026-08-25:
Yes, and in fact the octa style in my PR is a C++ implementation of BOSL2's octa. But that doesn't change the statement: BOSL2 isn't bundled, so beginners won't use it, so beginners get the crummy lat/lon spheres with no north or south poles.

MarkJeronimus, 2026-08-27:
> There is a decent argument that anything that can be done in userspace should be done in userspace

That argument breaks down in at least the following cases:
- The manual implementation is not trivial (in case of the octosphere, it's extremely complex and fiddly)
- The manual implementation is orders of magnitude slower (the case for the rounded cube or any type of chamfer/bevel in general)

stefano2734, 2026-09-20:
Label: Meshing ?
Label: Workaround? (by Bosl 2)
Other labels?

closed 2026-09-30 by jordanbrown0