openscad/openscad #1986: Feature Request: Concave Hull

ant-t, 2017-04-05:
Today I needed to cover the holes in an imported stl file, I ended up doing it by hand. It wasn't pretty or completely satisfactory. A concave hull would be an easy way to close internal holes of a geometry and add further functionality: http://ubicomp.algoritmi.uminho.pt/local/concavehull.html

doug-moen, 2017-04-05:
> uminho.pt/local/concavehull.html

This looks like something that can be implemented as an OpenSCAD library
module. Coding it in C++ isn't necessarily easier or a better solution.

thehans, 2017-04-05:
The links on that page to implementation are broken, there is no explanation of how the algorithm works.

MichaelAtOz, 2017-04-05:
Here you go [pdf](http://repositorium.sdum.uminho.pt/bitstream/1822/6429/1/ConcaveHull_ACM_MYS.pdf)
It's 2D.
@doug-moen how would you do it in OpenSCAD for a STL file?

MichaelAtOz, 2017-04-05:
I left my pure & applied maths behind at university, but [this](http://doc.cgal.org/latest/Alpha_shapes_2/index.html) speaks my language; 

> "As mentioned in Edelsbrunner's and Mücke's paper [2], one can intuitively think of an α-shape as the following. Imagine a huge mass of ice-cream making up the space R3 and containing the points as "hard" chocolate pieces. Using one of these sphere-formed ice-cream spoons we carve out all parts of the ice-cream block we can reach without bumping into chocolate pieces, thereby even carving out holes in the inside (e.g. parts not reachable by simply moving the spoon from the outside). We will eventually end up with a (not necessarily convex) object bounded by caps, arcs and points. If we now straighten all "round" faces to triangles and line segments, we have an intuitive description of what is called the α-shape of S. Here's an example for this process in 2D (where our ice-cream spoon is simply a circle):"

MichaelAtOz, 2017-04-05:
Here's the [3D CGAL alpha shapes](http://doc.cgal.org/latest/Alpha_shapes_3/index.html), I'm no CGAL expert, is this useful in an OpenSCAD context?

MichaelAtOz, 2017-04-05:
[Here](https://cran.r-project.org/web/packages/alphashape3d/alphashape3d.pdf) is a GPL algorithm in R

MichaelAtOz, 2017-04-05:
Paper [pdf](https://cran.r-project.org/web/packages/alphashape3d/vignettes/alphashape3d.pdf), link to [source/binaries](https://cran.r-project.org/web/packages/alphashape3d/index.html).
I'll stop now...

nophead, 2017-04-05:
Nice, although I had an alternative idea for concave hull(). Basically you
have a set of objects on the inside, as with the current hull(), but you
also have another set on the outside. The hull is like a rubber membrane
stretched over the inside objects with the outside objects pushed into it.
It is very useful for blending arcs, which can be done with convex hull
until you need a concave shape. The only way to do that now is by
calculating tangents.

GilesBathgate, 2017-12-16:
@MichaelAtOz The CGAL alpha shapes module as you suggested seems to be fit for this purpose.

I've thrown together a proof of concept implementation based on this: https://stackoverflow.com/questions/38829464/cgal-concave-hull-3d-alpha-shapes-3

It seems to work quite nicely, although I think it will probably need improvement:

[image]

hradec, 2024-01-14:
I'm running in this issue now. I need to create a hull of this cylinders: 
[image]

but using the normal `hull()` gives me this: 
[image]

So, as I guess no-one have done any work on this, right? So I'm willing to give it a try. 

I'll try to use the @GilesBathgate tip above and see what I can do. 

As I never tried to mess with openscad code before, it would be really helpfull if someone could point me to where the hull functionality is, and some documentation on how to add parameters to it. (I'm thinking on add a `concave=true` just the one on rapCAD above!)

wish me luck!