openscad/openscad #1988: difference() with union() CSG Normalization goes feral

MichaelAtOz, 2017-04-10:
See [Forum](http://forum.openscad.org/difference-module-operator-with-strange-behaviour-td21146.html).

```
difference()
{
    cube([10, 15, 6], center=true);
    //union()
        {
        translate([10, 0]) cylinder(r=2, h=20,  center=true);
        translate([-10, 0]) cylinder(r=2, h=20, center=true);  // works with translate([20, 0])
        }
 
} 
```

Uncomment the union() gets:
```
Compiling design (CSG Products normalization)...
WARNING: Normalized tree is growing past 800000000 elements. Aborting normalization. 
WARNING: CSG normalization resulted in an empty tree 
Normalized CSG tree has 0 elements
Compile and preview finished.
Total rendering time: 0 hours, 3 minutes, 42 seconds
```
Note that is after I doubled the elements in preferences.
#484 #319 may be similar.

woferry, 2017-04-10:
I'll just note that I could reproduce this on the Mac 2017.02.08 (git 2ef5333) build, but I could NOT reproduce on the released build, 2015.03.

kintel, 2017-04-10:
Looks like this regression happened sometime after 2015.09.14

kintel, 2017-05-14:
Git bisect points at this commit: 97cab4589bc995fdacc43383433040d3953b316b

commit 76262b17 (2017-05-15): Fixed bug introduced during refactoring. Fixes #1988
closed 2017-05-15 by kintel