openscad/openscad #1990: SVG and dxf export has incorrect dimensions for cicles in the y value

nonuniform, 2017-04-10:
Export tested with Win OpenSCAD 2015.03-2 and Win OpenSCAD 2017.01.20

Import tested with illustrator, inkscape, 3D Studio, Vetrics Aspire

A Cycle with r 1.5mm is after export x=3mm, y=2,994mm

I noticed that because i can not mill 3mm holes with my cnc router with an 3mm end mill.

t-paul, 2017-04-10:
Hmm, SVG should have been fixed via #1600 which would be in the dev snapshots.

nonuniform, 2017-04-10:
in Win OpenSCAD 2017.01.22? i don`t know how to run that snapshop. xD

With OpenSCAD 2017.01.20 x86 (64-bit) exe installer - i have tested it. The 3D Studio Max dxf importer can also fix the missing unit informations by transformation to mm. But cycle dimensions are also incorrect.

Maybe i had to wait for a 2017.01.22 exe installer. Thx for feedback

jepler, 2017-04-10:
This is most likely due to the way openscad divides primitives such as circles up into lines according to the special values `$fn`, `$fs`, and `$fa`.  As a simple remedy, you may want to try assigning `$fn` a value that is a multiple of 4 (e.g., `$fn=36`).

Can you show a minimal openscad file which demonstrates the problem? I tried with
````
circle(r=1.5, $fn=4);
````
and the generated svg has
````
<path d="
M 1.5,-0 L 9.18485e-17,-1.5 L -1.5,-1.83697e-16 L -2.75546e-16,1.5 z
" stroke="black" fill="lightgray" stroke-width="0.5"/>
````
As expected from the `$fn` value, this is a 4-sided circle, and the errors introduced by the limits of floating point precision are down in the range of 1e-16.  Importing this file into inkscape, I set the stroke width to 0, and the axis-aligned bounding box size is reported as 3.000mm×3.000mm.

Other `$fn` values give different reported bounding-box sizes, because they subdivide the circle differently.  For instance, `$fn=14` gives 3.000×2.925, because the approximation of the circle has points on the ±X axis but flats on the ±Y axis.  Circles with `$fn` values that are multiples of 4 give poins on both axes (until they're rotated, anyway)  For more about these special values, see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#.24fa.2C_.24fs_and_.24fn

t-paul, 2017-04-10:
Ah, thanks @jepler, good point. I guess that's a good candidate for the FAQ page.

nonuniform, 2017-04-10:
i noticed the problem by this Openscad Project https://github.com/lautr3k/RepRap-iTopie/tree/dev/scad

For me i fixed it, by expanding the radius to: m3_screw_radius to 1.504; 

i have finaly used the dfx export, because of the correct units by importing to cam Software, vetrics aspire. SVG had the wrong size, i think because of missing mm.

nonuniform, 2017-04-10:
in my project the fn value is 50, looks high, maybe i will change it to a higher value multiples by 4, but i think i will always have the same problem with my 3mm end mill, because it`s just a gegon. Only bigger holes can fix it, or circles based on bezier corners, alway the best solution. Thx for you help!

jordanbrown0, 2024-01-16:
Perhaps the $fa/$fs calculation should round up to a multiple of 4, so that there is always a vertex on each axis.  Perhaps it should do this only when the result is, say, >12, so that it's an almost imperceptible change.  (There's a lot of difference between a 5-gon and an 8-gon, but not a lot between a 13-gon and a 16-gon.)

nophead, 2024-01-16:
I have a circle4n() module and r2sides4n() function that does that in NopSCADlib.