openscad/openscad #1993: Issue with font cache dialog

chriscamacho, 2017-04-14:
When starting OpenSCAD specifically with a source file that includes the text command I will intermittently see an empty (unrendered) dialog.

I have seen this issue with two Void Linux systems both with working fontconfig installs

system fonts do work normally with OpenSCAD despite this

please see this link for more information http://forum.openscad.org/small-see-through-dialog-when-starting-td21144.html

matthew-viglione, 2018-03-07:
I get this same issue. I also think it's the font cache updating, but leaving it open for a long time doesn't make it go away. (I would think that it should disappear once the font cache is rebuilt). Anyone else see this? 

[image]

chriscamacho, 2018-03-07:
...well... at least its not just me, fortunately it hasn't stopped me doing stuff (none modal!) I haven't experienced any particular font issues either, have you?

matthew-viglione, 2018-03-07:
I haven't. I did more digging on this issue and saw that other people have had it too. I'll try to find links to the forums where this has been discussed and post it here. 

The general consensus is that it is the notification window that the font cache is being updated, and that it doesn't render properly because of missing libraries. I would think that it would go away after a while, like it does on my Windows computer that did the same thing (although the small window did render properly on the Windows machine, and was the font cache updater).

MichaelAtOz, 2018-03-08:
See #1892 the bit about deleting the fontconfig folder, the above pop-up window may be a leftover from some timing issue.

matthew-viglione, 2018-03-08:
Where are these folders located on Linux? I've read through the fontconfig man pages, and flushed and reset to fontconfig cache manually, but this doesn't resolve the problem.