openscad/openscad #1994: Replace tab with 2 spaces in CSG Tree display

MichaelAtOz, 2017-04-14:
With a complex model the tree indents way too much.

kintel, 2017-04-15:
For display-only, should be able to do this using Qt.
For export, we may want to look into all other indentations, which will touch a lot of the test cases.

ChrisVesters, 2017-04-16:
Do we want to simply change this? Or do we want to use some configuration as well (maybe the same as for entering yourself)?

ChrisVesters, 2017-04-16:
It is easy to lower the tab stop width in Qt. If we want to make it exactly 2 spaces, then we will have to do some calculations with the font and stuff (which might be overkill...)

Also, just replacing the size for the CSG Tree, makes it a bit inconsistent with the AST. I think we should keep both of them in sync.

kintel, 2017-04-16:
I'd accept a display-only fix.

MichaelAtOz, 2017-05-29:
Thanks Chris

closed 2017-05-29 by MichaelAtOz