openscad/openscad #1997: stl exporting

guowcas, 2017-04-18:
as long as i see, OpenSCAD only exports .stl mesh in txt format. Is that correct? 
is there any plan to enable OpenSCAD to export binary .stl mesh, so that features like colour can be included? 
thanks a lot.

t-paul, 2017-04-18:
Yes, currently only ASCII STL is supported for export. Issue to extend the export options is #649.

Binary STL will not really help with color though.

closed 2017-04-18 by t-paul

guowcas, 2017-04-18:
thanks a lot. 

Dr. GUO Wei
Reactor Physics Department, 
Shanghai Institute of Applied Physics, CAS

MichaelAtOz, 2017-04-18:
@guowcas re colour, see #1608 for relevant discussion.
However OpenSCAD2 appears to be quiescent (or worse), while the concept wasn't directly fixed to OpenSCAD2, I think Doug has focus elsewhere ATM. 
As a discussion for a specification it is worth reading. Perhaps consider whether such approaches would meet whatever requirement you have.