openscad/openscad #1999: feature reeust: auto reload-to-generate-stl

rewolff, 2017-04-22:
Just like there is an automatic reload-and-preview, I would like to have a feature where I can set openscad to automatically reload-and-render, even to the level: automatic regenerate and overwrite the STL file..... 

I see that in my current workflow, I have pretty much finished my design and I change minor things in my source and then want to re-export to STL to slice and print the design. So while reload-and-preview is great to see the changes, I now see myself mouse over to openscad, hit F6, when that finishes, mouse to the file menu, export -> STL -> accept filename, -> confirm overwrite. 

I then need to reload the design in slic3r. I'm told that has been improved in the current version.

rewolff, 2017-04-22:
Party like it's 1999!

t-paul, 2017-04-22:
I don't see how that fits well into the GUI model without causing major issues for other scenarios. Reload and preview already has some issues but solves a workflow that actually requires the GUI to be shown.

Automatic export might work much better into the command line model or maybe using some automation flow integrating the tools.

rewolff, 2017-04-22:
Hmm. That solves it "for me".... I'm good at commandlines. I do think that others might benefit from an "automate this recurring sequence of menu selections". 

For myself, I've written a program called "autodo" that issues a command when the named file changes. 
`% autodo printjeshouder.scad openscad -o printjeshouder.stl printjeshouder.scad
`
indeed does what I want. Thanks for pointing me in the right direction.... :-)

MichaelAtOz, 2017-04-22:
Similar requests, for various auto-workflow stuff, come up from time to time.
Perhaps we could have a very basic, run a script command, shortcut ctl-shift-R.
Basically a text field in parameters, with simple substition of the filename (minus extention). May need the current directory too, I suspect on Windows the cmd will open in the current directory(?).
Using rewolff's example,
(for Windows) 
`cmd.exe /C "openscad -o @filename@.stl @filename@.scad" && "someslicer @filename@.stl"`

Yes, it could open a can of worms...

kintel, 2017-04-23:
A cmd-line interface like this sounds a bit like baby steps towards UI scripting. There are various avenues towards implementing UI scripting, some platform-specific ones (like AppleScript), framework-centric ones (QtScript) or roll our own (e.g. Lua bindings). Given that OpenSCAD is in itself a language, self-hosted scripting could also makes sense, although that could quickly get confusing..
A simple cmd-line interface supporting simple variable replacement could be a low hanging fruit though, given the limited resource situation.

PRouzeau, 2017-04-23:
There is definitely needs to automate STL/DXF exports.
There was also requests to export multiple stl/dxf from one scad file according to a parameter (typically ‘part’ number)
To help that process, I created a libre office calc sheet which does create command line batch files capable to export a full set of parts (STL and DXFs) in one go, from one or multiple scad files. 
However, I have to enter scad file name, part names and references manually in my LO sheet.
So, it does work in two phases :
- If I change/add new parts, I have to modify my LO sheet and regenerate my batch file(s) 
- If I modified parts without adding/removing one, I only run the existing batch files to recreate full sets of parts.
This have helped a lot my workflow.
I have published an example of this batch generator (Windows only) here :
https://github.com/PRouzeau/Garden-train-Monorail/blob/master/Win_batch_Monorail.ods
This is an old version, the more recent being capable to also export dxf panels.
It is associated with a part selector within the openScad file(s).
By default this OpenSCAD part selector will display the ensemble if the variable part=0 (or not existing).

However, you have to manage a different batch generator for each project, so I think it can be better to set the names and batch ensemble in the scad file itself, in the form of an array, or array of arrays
which could have the form of: 
parts = [["myscad.scad", "ensemble", 0], ["myscad.scad", "part 1 name", 1], ["myscad.scad" , "part 3 name", 3], ["othercad.scad" , "part 7 name", 7]];
But yet this won’t work as you cannot within a command line name an export file from reading an internal OpenScad variable. Maybe creating a standard variable for export name ?

rewolff, 2017-04-23:
For "anmimation" we already have a mechanism where openscad iterates a variable, and can do specific things, in that case "export image". 

How about an option to iterate to N and call the "module export (n)", implemented in the file, which can then instantiate whatever is necessary for  subassembly n, and then trigger a render, and export. 
This is partly the "we already have a language" idea mentioned above.

kintel, 2017-04-23:
@rewolff I think that the UI feature for scripting/export is separate from any language feature for supporting export of subassemblies. See e.g. #1608 for one of the discussion on language support.

nophead, 2017-04-23:
I manage to automate all my stuff with Python and echo(). I use echo to
generate a structured BOM and from that I know what modules to call to
generate all the STL files, etc. I don't need any support from OpenSCAD
other than command line and echo.

MichaelAtOz, 2017-04-23:
@nophead presumably using the .echo command line output (I only noticed that recently)?
Has anyone used the command line make?

nophead, 2017-04-23:
No my code predates ```.echo```. I just capture stdout. I should probably update it to use ```.echo```.

PRouzeau, 2017-04-25:
So, instead of using Libre Office to create my export batch files (Windows), I created them as ‘.echo‘ files with OpenScad, which make the process a bit more simple:

One set of of parts to be exported is a vector of vectors.
One vector for a part is:
- 1rst parameter of a part is its number (as used by the part selector): Compulsory parameter
- 2nd parameter is the file name WITH extension (so this allow exportation of dxf, png, echo). Optional parameter, default format will be stl
- 3rd parameter is the scad file, so you can call multiple scad files in one part set, Optional parameter, default scad file will be used.

There are defaults variables for scad file and default export directory is «STL» (export directories shall exists, as they are not created by the batch) You can change the destination directory as a parameter in the batch creation module. Beware that openSCAD is case sensitive.

In its simplest form (exporting stl parts without name, only number), an example of set vector could be 
```
set1 = [[1],[3],[7]];
```
a more sophisticated part set could be 
```
set2 = [[1,"part1_name.stl"],[3,"part3_name.stl"], [1,"part1_name.png"],
       [7,"part7_name.png"],[11,"part11_name.stl","otherscad.scad"]];
```

batch echo output is also considered as a part in the part selector. I choose part 201 and above.

You first shall create a batch generator file ‘create_batch.bat’ with as many lines as you define part sets:
```
start c:"\Program Files\OpenSCAD\openscad.exe" -o batch_set1.echo -D part=201 myscad.scad
start c:"\Program Files\OpenSCAD\openscad.exe" -o batch_set2.echo -D part=202 myscad.scad
cmd
```
Run this generator then change the extension of your created batch files from ‘.echo’ to ‘.bat’

The lines and modules in you openscad file are :
```
part selector….
else if (part==201) {
  batch(set1);
}
else if (part==202) {
  batch(set2, PNG); //example to impose another output directory than ‘STL’
}

thisScad = "myscad.scad"; // Default scad file for batch
set1 = [[1],[3], [7]]; // this is an example
set2 = [[1,"part1_name.png"],[3,"part3_name.png"],[7,"part7_name.png"]]; 
  // another example with image export

module batch(set, directory="STL") {
  bt1 = str("\nstart c:\"/Program Files/OpenSCAD/openscad.exe\" -o ",
            (directory)?str(directory,"/"):"","myproject_");
  function line (val) = 
        str(bt1,(val[0]<10)?"0":"",val[0],"_",(val[1])?val[1]:".stl",
             " -D part=",val[0]," ", (val[2])?val[2]:thisScad,"\necho: "); 
  
   for (i=[0:len(set)-1])
     echo (line(set[i]));
   echo("\ncmd\necho: "); // to maintain command window open (to have a look on messages)
}
```
In the batch, you cannot remove ECHO strings, but  the «ECHO: "» commands being alone in their lines, they will only output double quote, which is minor trouble.

Note that in the Windows batch I shall give the full path to OpenSCAD program, else that won't work. This means the batch shall be adapted for a given machine.

kintel, 2017-04-25:
You should be able to add the OpenSCAD install folder to you PATH variable to avoid specifying the full path.

MichaelAtOz, 2017-04-25:
On windows it is added to the PATH if you use the installer download rather than the zip file.
(I tend to install the actual releases, and use zip for side 'installs' of snapshots)

oysteinkrog, 2017-04-25:
I use a makefile to automate my build process, and define all my parts inside .scad files.
https://github.com/oysteinkrog/kmax1/blob/master/Makefile

My makefile looks for all .scad files, then scans them for modules that has the "part_*()" pattern, and builds those modules to stl output files.
It builds upon the standardized makefile provided in the wiki, but is a bit more advanced.
Here is an example of a file with a part in it:
https://github.com/oysteinkrog/kmax1/blob/master/y-carriage-bearing-mount.scad
Or this one which has multiple parts:
https://github.com/oysteinkrog/kmax1/blob/master/x-carriage.scad

I simply run "make all" whenever I make a changed.
Only what needs to be rebuild will be, not all the parts.

If one wants to automate the "make all" part, simply use a file-watcher utility and have it run make when a .scad file changes.

PRouzeau, 2017-04-26:
There is another important reason to automate part export, this is the new functionality Customizer and its datasets, so I extended my batch generator to be able to export parts associated with a Customizer dataset (if defined):
```
thisScad = "myscad.scad"; // Default scad file for batch
thisProject = "myproject"; //project name set in parts files
thisDataset = "Dataset1"; //variable defined only if using a customizer dataset
thisJSON = str(_join([for (i=[0:len(thisScad)-5]) thisScad[i]], len(thisScad)-5,""),"json");

module batch(set, directory="STL") {
  bt1 = str("\nstart c:\"/Program Files/OpenSCAD/openscad.exe\" -o ",
(directory)?str(directory,"/"):"",thisProject,"_",(thisDataset)?str(thisDataset,"_"):"");
  function line (val) = 
    str(bt1,(val[0]<10)?"0":"",val[0],"_",(val[1])?val[1]:".stl",
    (thisDataset)? str(" --enable=customizer -p ",thisJSON," -P ",thisDataset," "):"", 
    " -D part=",val[0]," ",  
  (val[2])?val[2]:thisScad,"\necho: "); 
  
   for (i=[0:len(set)-1])
     echo (line(set[i]));
   echo("\ncmd\necho: "); // to maintain command window open (to have a look on messages)
}
// function extracted from relativity/string library by davidson16807
function _join(strings, index, delimeter) = 
  index==0 ? 
    strings[index] 
  : str(_join(strings, index-1, delimeter), delimeter, strings[index]) ;
```

Unfortunately, as for version 2017.01.20, there is a problem in the command line if a variable is defined with ‘-D variable=value’, it don’t supersede the value of the same variable as defined in the customizer dataset (I think it should).
So, the generator cannot select the part which is stuck to the value defined in the dataset.
To have the batch generator working, I had to remove manually in the ‘.json’ file in the dataset(s) the variable "part":"0". This cause no trouble, but add a step in the process.

PRouzeau, 2017-05-22:
I refined a bit my batch generator and added it in my OpenSCAD library here : 
https://github.com/PRouzeau/OpenScad-Library

rewolff, 2026-02-26:
OK. Great! Thanks for the clarification!