openscad/openscad #2001: Issues found by PVS-Studio

VasilievSerg, 2017-04-24:
Hello,
I have found a few issues using PVS-Studio analyzer. PVS-Studio is a static code analyzer for C, C++ and C#: [PVS-Studio on Visual Studio Marketplace](https://marketplace.visualstudio.com/items?itemName=EvgeniyRyzhkov.PVS-Studio)

Some of the analyzer warnings:
- V523 The 'then' statement is equivalent to the 'else' statement. /src/lodepng.cpp:5142
[A link to the source code](https://github.com/openscad/openscad/blob/0a00aecc6f3a510a8ef0c396bc04a73e5746cbeb/src/lodepng.cpp#L5142)
_Regardless of the value of the "prevline' variable the same actions will be performed, as both of the 'if' branches have identical expressions. Perhaps, there is a logical error here._

- V640 The code's operational logic does not correspond with its formatting. The second statement will always be executed. It is possible that curly brackets are missing. src/OGL_helper.h:620
[A link to the source code](https://github.com/openscad/openscad/blob/0a00aecc6f3a510a8ef0c396bc04a73e5746cbeb/src/OGL_helper.h#L620)
_In the loop only one statement "f->debug();" will be executed. The statement "os << std::endl;" is not related to the loop. Perhaps, curly brackets are missing here._

- V623 Consider inspecting the '?:' operator. A temporary object of the 'QByteArray' type is being created and subsequently destroyed. Check second operand. src/mainwin.cc:1784
[A link to the source code](https://github.com/openscad/openscad/blob/0a00aecc6f3a510a8ef0c396bc04a73e5746cbeb/src/mainwin.cc#L1784) 
_The [documentation](https://www.viva64.com/en/w/V623/) to the diagnostic rule gives more details about this issue._

In addition, I suggest having a look at the emails, sent from @pvs-studio.com.

Best regards,
Sergey Vasiliev

kintel, 2017-04-24:
Thanks! The last one looks accurate. The first two are actually 3rd party software and would best be reported upstream.

kintel, 2017-09-09:
Closing as automated marketing

closed 2017-09-09 by kintel