openscad/openscad #2005: Recursion checks fail when built with MSYS2

nophead, 2017-04-25:
Recent Windows snapshots correctly detect recursion but when built with MSYS2 and run on Win7 64bit the openscad.exe terminates with exit code C00000FD, which is stack overflow.

t-paul, 2017-04-25:
On Windows we currently [assume a fixed stack size](https://github.com/openscad/openscad/blob/master/src/PlatformUtils-win.cc#L100) which is set in [mingw-cross-env.pri](https://github.com/openscad/openscad/blob/master/mingw-cross-env.pri#L38).

nophead, 2017-04-25:
Looks like that doesn't get passed when doing a MSYS2 build. I.e. that section isn't active in the qmake.

nophead, 2017-04-25:
I added it to the linker options in a win* section and that fixes the MSYS2 build. I don't see any point in adding it to the QMAKE_CXXFLAGS as it is only the linker that needs to know.

```
win* {
  RC_FILE = openscad_win32.rc
  QMAKE_CXXFLAGS += -DNOGDI
  WINSTACKSIZE = 8388608 # 8MB # github issue 116
  QMAKE_LFLAGS += -Wl,--stack,$$WINSTACKSIZE
}
```

Should I also remove it from mingw-cross-env.pri? I.e. are win* sections active when cross compiling?

Since it needs to match  the value used in STACK_LIMIT_DEFAULT I think that WINSTACKSIZE should be passed as a -D definition from $$WINSTACKSIZE in openscad.pro.

kintel, 2017-04-26:
@nophead I think they can be moved to `win*` somewhere. `win*` must be defined, also when cross-compiling. Otherwise some of the source files specified in `openscad.pro` wouldn't be compiled.

nophead, 2017-04-28:
Fixed by #2010

closed 2017-04-28 by nophead