openscad/openscad #2006: Dump tests fail on Windows due to extra leading zero in exponents

nophead, 2017-04-25:
```
-square(size = [1e+06, 1e+06], center = true);
+square(size = [1e+006, 1e+006], center = true);
```
This is true both for Windows snapshots and MSYS2 builds when run on Win7 64 bit machine.

kintel, 2017-04-26:
There is a python function in `test_cmdline_tool.py` which normalizes floating point output across platforms: https://github.com/openscad/openscad/blob/master/tests/test_cmdline_tool.py#L108

This function doesn't normalize differences in formatting of scientific notation. The correct fix would be to add that, i.e. remove double leading zeros when possible.

nophead, 2017-04-26:
I think the function would probably normalise them because Python2 prints two digit exponents on my system. The problem is the numbers that are failing are 1e+006, so the function isn't even called because the regex only matches numbers with a decimal point.

I will see if I can fix the regex.

kintel, 2017-04-28:
Fixed by #2009

closed 2017-04-28 by kintel