openscad/openscad #2007: cgalstlsanitytest_normal-nan test fails on Windows

nophead, 2017-04-25:
It seems to try to run a Python program as a Win32 executable.
```
cgalstlsanitytest_normal-nan

882/1084 Testing: cgalstlsanitytest_normal-nan
882/1084 Test: cgalstlsanitytest_normal-nan
Command: "C:/msys64/mingw64/bin/python.exe" "C:/msys64/home/ChrisP/openscad/tests/test_cmdline_tool.py" "--comparator=" "-c" "C:/msys64/mingw64/bin/convert.exe" "-s" "txt" "-t" "cgalstlsanitytest" "-f" "normal-nan" "C:/msys64/home/ChrisP/openscad/tests/cgalstlsanitytest" "C:/msys64/home/ChrisP/openscad/tests/../testdata/scad/misc/normal-nan.scad" "C:/msys64/home/ChrisP/openscad/tests/../Release/openscad.exe"
Directory: C:/msys64/home/ChrisP/openscad/tests
"cgalstlsanitytest_normal-nan" start time: Apr 25 17:29 GMT Daylight Time
Output:
----------------------------------------------------------
run_test() cmdline: ['C:/msys64/home/ChrisP/openscad/tests/cgalstlsanitytest', 'C:/msys64/home/ChrisP/openscad/tests/../testdata/scad/misc/normal-nan.scad', 'C:/msys64/home/ChrisP/openscad/tests/../Release/openscad.exe', 'C:/msys64/home/ChrisP/openscad/tests/cgalstlsanitytest-output/normal-nan-actual.txt']
using font directory: C:/msys64/home/ChrisP/openscad/tests/testdata
normal-nan
Error: %1 is not a valid Win32 application "C:/msys64/home/ChrisP/openscad/tests/cgalstlsanitytest"
<end of output>
Test time =   0.06 sec
----------------------------------------------------------
Test Failed.
"cgalstlsanitytest_normal-nan" end time: Apr 25 17:29 GMT Daylight Time
"cgalstlsanitytest_normal-nan" time elapsed: 00:00:00
```

kintel, 2017-04-26:
Right, unix-style shebangs probably doesn't work on Windows.

kintel, 2017-04-26:
Could you check if this does the trick?
```
diff --git a/tests/CMakeLists.txt b/tests/CMakeLists.txt
index 499f0dd..3d5370d 100644
--- a/tests/CMakeLists.txt
+++ b/tests/CMakeLists.txt
@@ -1642,7 +1642,7 @@ add_cmdline_test(csgpngtest EXE ${PYTHON_EXECUTABLE} SCRIPT ${CMAKE_SOURCE_DIR}/
 add_cmdline_test(throwntogethertest EXE ${OPENSCAD_BINPATH} ARGS --preview=throwntogether -o SUFFIX png FILES ${THROWNTOGETHERTEST_FILES})
 # FIXME: We don't actually need to compare the output of cgalstlsanitytest
 # with anything. It's self-contained and returns != 0 on error
-add_cmdline_test(cgalstlsanitytest EXE ${CMAKE_SOURCE_DIR}/cgalstlsanitytest SUFFIX txt ARGS ${OPENSCAD_BINPATH} FILES ${CGALSTLSANITYTEST_FILES})
+add_cmdline_test(cgalstlsanitytest EXE ${PYTHON_EXECUTABLE} SCRIPT ${CMAKE_SOURCE_DIR}/cgalstlsanitytest SUFFIX txt ARGS ${OPENSCAD_BINPATH} FILES ${CGALSTLSANITYTEST_FILES})
 
 #
 # Trivial Export/Import files
```

nophead, 2017-04-26:
Yes that works.

Shebangs only work on Windows inside a Cygwin shell. To make a Python file
that works in native Windows and Unix it needs to have both a .py extension
and the shebang. I don't know if it would work in popen though.

nophead, 2017-04-28:
Fixed by #2008.

closed 2017-04-28 by nophead