openscad/openscad #2011: [Enhancement] integrate "astyle" into Openscad

thawkins, 2017-04-27:
It would be great to have a menu-entry that would push the scad source though "astyle" with suitable or definable parameters. 

Astyle is a source for matter for "C like/ {} block " languages, astyle -A2 xxx.scad does wonders for readability of scad files, it would good to have a "reformat code" menu-entry. 

see: https://github.com/live-clones/astyle/tree/master/AStyle

t-paul, 2017-04-27:
See #1151, the current view is that it's not going to be the best strategy to get the reformatting feature.

thawkins, 2017-04-27:
When is the correct stratergy planned for? That ticket is almost 2 and half
years old, and basicaly comes to no conclusion.

Perhaps a streaming plugin that allows the source to be streamed through
any designated external tool. That would allow other tools like validators,
custome exporters etc also to be used.

t-paul, 2017-04-27:
One of the problems is that OpenSCAD is not C, just looks similar, and the snapshot versions now have even more constructs that are very different to C/C++. With the limited dev resources it's difficult trying to manage an additional parser just for the reformatting.
As there are lots of requests for better integration of source code and visualization, one option would be to use the already existing parse tree to do the reformatting.