openscad/openscad #2014: OpenSCAD freezes when zooming in

thehans, 2017-04-29:
Steps to reproduce:
1) Open OpenSCAD
2) mousewheel zoom in for a little bit, maybe about one order of magnitude
The application becomes unresponsive and requires the process to be killed.

2017.04.29.nightly (git a322fc8) 
Linux Mint 18.1 w/ Cinnamon Desktop 
NVIDIA Corporation GM107GLM [Quadro M1000M] (rev a2)

t-paul, 2017-04-29:
Sounds like https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=858904

thehans, 2017-04-29:
It happens exactly when the distance of the next frame would be less than 10.  

It also happens when zooming out past a certain point.  Attached is a screenshot of where it froze from zooming out.
[image]

This issue is not present in 2015.03-1, the only other version I have installed.   I think this behavior is from a very recent change, It seems like I would have noticed it pretty quickly.

t-paul, 2017-04-29:
Hmm, is that with the proprietary driver or with nouveau? Anything in `dmesg` from the driver?

t-paul, 2017-04-29:
Ok, I suspect a bug in the number display. I can reproduce it with scale numbers on, but not if those are disabled.

thehans, 2017-04-30:
Same here, disabling scale markers works around the issue.

I am using nvidia proprietary drivers by the way

kintel, 2017-04-30:
Idea for debugging:
* Look at https://github.com/openscad/openscad/blob/master/src/GLView.cc#L507
* Comment out all `gl*` function calls -> verify that it fixes the problem
* Replace all `gl*` function calls with `printf`'s -> inspect the output looking for trouble
* If still necessary: Call all `gl*` functions with the `printf` output, verify that it still fails and post here.

t-paul, 2017-05-01:
It's the loop in [`showScalemarkers()`](https://github.com/openscad/openscad/blob/master/src/GLView.cc#L507). Needs a closer look to find out which parameters turn it into an endless loop...

kintel, 2017-05-01:
I can reproduce on OS X as well. Will look at it.

commit 85b015e3 (2017-05-01): Regression: j variable got wrong type during refactoring. Fixes #2014
closed 2017-05-01 by kintel