openscad/openscad #2015: Add command to center something.

Harvie, 2017-04-30:
Hello,
i think it would be useful to have some simple command to center any object in OpenSCAD.

eg. if i have `translate([10,10,10]) cube();`
i'd like to center it back using `center() translate([10,10,10]) cube();`
(it surely does not make any sense to translate something only to center it back. However there are use cases for this. Centering solids produced by 3rd party subroutines, or centering import()ed objects that are not centered in original files.

It's fairly easy. Complicated objects can be simply treated like smallest possible cube/block that fits around that object. Such block is then easy to center.

Also there can be something like `fit_possitive()` and `fit_negative()` that would corner whole object next to the center of coordinate system, but completely on positive or negative side. That makes a lot of sense for CNC / CAM purposes, because eg. for milling you want to have everything in negative coordinates (you are milling down to material from origin), while for 3D printing you usualy use possitive coordinates (you are adding material up from origin).

It would also make sense to be able to use all of these only for one or two axes, while leaving rest intact.

kintel, 2017-04-30:
See #586 for a discussion on this topic. This looks like a duplicate of #586

oysteinkrog, 2017-04-30:
Since this is not currently available in openscad, I've landed on using this syntax:
cubea(size=[10,10,10], align=X-Y);
cylindera(r=5, h=20, orient=Y, align=-Y);

Hopefully it is self-explanatory.
X, Y etc are simply global variables for [1,0,0] and [0,1,0].

https://github.com/oysteinkrog/thing_libutils/blob/master/shapes.scad

MichaelPFrey, 2018-10-27:
see also #1610

jordanbrown0, 2024-11-23:
Duplicate of other reports.

closed 2024-11-23 by jordanbrown0