openscad/openscad #2016: Can't use boolean operations on imported STL (e-incident_sface() != SFace_const_handle())

Harvie, 2017-04-30:
I've took `Dremel_Mount_Assembled.stl` from http://www.thingiverse.com/thing:600158 and imported it like this:

```
difference() {
    import("Dremel_Mount_Assembled.stl", convexity=5);
    cube(150);
}
```

However when i hit F6 it gives me this error:

```
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
CGAL Cache insert: difference(){import(file="Dremel_Mount_A (0 bytes)
CGAL Cache insert: group(){difference(){import(file="Dremel (0 bytes)
Geometries in cache: 10
Geometry cache size in bytes: 3871952
CGAL Polyhedrons in cache: 6
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 0 seconds
Rendering finished.
```

When i hit F5 it cames out broken:

[image]

besides of difference() it also has similar problems with union() and intersection()...

kintel, 2017-04-30:
This is likely a duplicate of #1824

t-paul, 2017-04-30:
I don't get the strange display issues here, so please add the system and version information as mentioned in the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#How_do_I_report_bugs.3F).

Otherwise the STL seems to have zero-faces (as reported by the Blender check) that OpenSCAD/CGAL can't handle yet. Possible ways to [fix the model](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6.3F) are also listed in the FAQ.

Harvie, 2017-04-30:
@kintel you are true. i was searching, but probably not hard enough. I will close this.
@t-paul it's latest official release (= 2015.03-3)

closed 2017-04-30 by Harvie

t-paul, 2017-04-30:
In case it's on Windows, maybe check if you have the latest graphics driver. That strange F5 output is probably unrelated to the zero-faces problem happening with F6-render.