openscad/openscad #2017: CAM functionality

Harvie, 2017-04-30:
Hello,
i would be happy if i could prepare operations for CNC milling directly in OpenSCAD code.

Currently i have to export stl and import it into some software like [PyCAM](http://pycam.sourceforge.net/). But the PyCAM does not really understand the different areas of object and treats whole model the same way. But if you want to achieve good surface finish and acuracy, then you need to treat each aspect of part differently. Eg. if you have cube with hemisphere on top, you should mill the hemisphere using so called surfacing operation and then mill the cube using profiling operation. However PyCAM takes STL and batch-processes it, so whole part is both surfaced and profiled. That makes the resulting g-code very slow and inefficient.

So i would like to have some way to generate g-code directly from OpenSCAD using some additional hints in .scad file that would define what should be profiled, drilled, pocketed, surfaced, etc...

I know that CAM operations are quite specific for every kind of production and you need completely different approach for milling on various types of machines, 3d-printing (CAM operations for 3D printing are obviously more straightforward). So this should be heavily modular system using plugins keeping openscad clean of hardware specific code.

However it would be nice if such CAM plugins could add another overlay layer that would render toolpaths directly in OpenSCAD window. And make some operations on openscad objects that would result in generating such code.

Everything should be done in way that does not interfere with normal openscad operation. So you will have .scad file and you will be able export it to STL completely ignoring all CAM hints in code. Or you will be able to export the g-code.

How i imagine this stuff working?

For example let's say you will have shape like this:

```
difference() {
    cube(100, center=true);
    translate([0,0,50]) sphere(40);
}
```

[image]

You can then add some CAM notations:

```
//load cam plugin for 3-axis milling
cam_load("mill_3-axis");

//define some speeds/feedrates/etc...
cam_rates(300,100,1000);

//define shape of bit that will be used for milling (might or might not be openscad object)
cam_mill_bit() { cylinder(20,3.175/2,3.175/2); }

difference() {
    //add operation to mill that cube from outside
    cam_mill(profile, outside) cube(100, center=true);
    //add operation to surface inside of sphere (because the sphere is removed - may be autodetected from difference())
    cam_mill(surface, inside) translate([0,0,50]) sphere(40);
}
```

And that's all. Generating milling operations (and other kinds of g-code) can be that easy if done right! Please think about this.

t-paul, 2017-04-30:
This sounds like a request for a totally new program.

Harvie, 2017-04-30:
@t-paul not really. when you think about it... These new functions can be completely transparent to standart workflow. They will do nothing (eg. like `translate([0,0,0])`). but internally they will be able to pass these individual objects to that CAM plugin which will in turn be able to generate toolpaths (g-code) for these primitives. And OpenSCAD will be able to treat these toolpaths as another optional layer in viewport.

That's basically all what OpenSCAD needs to do once user request CAM operations to be generated:

1.) load CAM plugin of users choice
2.) pass parameters (speeds, tool type) to this plugin (without any modification or action, just read from code and pass)
3.) pass individual primitives with CAM parameters to this plugin
4.) take g-code back from plugin and render it, so user can overview it (it's very simple. basicaly just lines and arcs that are connecting known coordinates)
5.) allow user to save that g-code to file

This sounds very doable to me!

Harvie, 2017-04-30:
The plugins can then be implemented as PyCAM based script that will take STL for that individual object, process it and return g-code... resulting g-code will be just concatenation of all produced g-code fragments. However they have to be sorted in way that makes sense. But user can do this by specifing order as integer for each operation.

You can even implement 3d priting slicer plugin. But in that case all you need is to process whole model and there's no real need to do that directly from OpenSCAD. However it might be handy in some cases.

Note that in this concept the generating of g-code itself is never done in OpenSCAD itself. Because there would be no point in that. All you need is to allow some external CAM program/plugin to process individual areas of your design in different manner. Rather than processing whole STL as single piece.

Neon22, 2017-04-30:
Sounds like this should be in the forum maybe for a wider discussion.
But I can see how you might - within openSCAD - have the scad program, the resulting STL, the sliced STL (courtesy of a slicer run as a sep process like most of them do).

Then within openscad:
- translate and project at the same layer height as the slicer (resultnig in a slice)
- compare the solids you're hitting on the project with the output gcode, and
- annotate/modify the gcode based on where you're hitting it and some kind of CAM knowledge.

Not sure if that's actually the right thing to do. But this could be done as a wrapper on top of openscad and so not require any changes to openSCAD specifically - except possibly to determine which primitives are being hit by the project.

My 2c...

kintel, 2017-04-30:
RapCAD original planned to include CAM. Perhaps there are some ideas there: https://gilesbathgate.com/2011/07/08/how-rapcad-will-do-cam/

nophead, 2017-04-30:
Seems to me a bad idea to combine CAD and CAM programs just to get around deficiencies in a particular CAM program.  Especially when OpenSCAD has limited developer resources it shouldn't try to everything.

Plus I am not sure PyCAM can't do what you want anyway. Each task can have its own bounds definition so I think you can limit the surface and profile operation to where you want them. It is pretty fundamental to milling to be able to do that.

Harvie, 2017-04-30:
@nophead 
> Each task can have its own bounds definition so I think you can limit the surface and profile operation to where you want them.

You are true. In PyCAM you can use bound box, which is not really that usefull for complex shapes.
Eg. for that cube with spherical hole inside i've demonstrated earlier:

You can limit surfacing operation just to that spherical hole, but once you are going to profile cut the cube from stock, you can't prevent it from milling the sphere again (even when there's no material = useless waste of time). That's simply because the sphere lays inside bounding box of the cube and therefore can't be de-selected using such simple approach (without really understanding geometry model, which is what OpenSCAD understands, but PyCAM doesnt).

nophead, 2017-04-30:
Well that is just a bug / deficiency in PyCAM then because there are many CAM tools for milling that are separate programs from the many 3D CAD tools. They must allow simple jobs like that to be milled efficiently.

Harvie, 2017-05-01:
@nophead in fact it does not matter what PyCAM does or wheter there are bugs. Lot's of CAD programs have CAM plugins, so you have everything integrated in one interface. I believe this can be done OpenSCAD way and i think it would be elegant as hell if done right. Because there's so much elegance in using OpenSCAD. I am truly amazed how simple and effective it is to learn and use. And i just want to apply the same elegant approach to manufacturing field.

To show you what i mean. I started working on small openscad script that will be able to import, simulate and reverse-engineer the g-code directly in OpenSCAD.
Idea is really simple: It takes any openscad object and moves it along the path defined by g-code. If this object is a little sphere you can use it for highlighting toolpaths or making SCAD model from 3d printer g-code (set sphere diameter to diameter of extruder). If this object has same shape as endmill, you can use it to create SCAD model from g-code meant for milling. In such case you obviously have to use difference() to remove produced shape from another object which is representing the stock material to be milled.

I am not sure if there's better way to do this, but i simply translate the object from one place to another and then make the path from that using hull(). Here's the WIP code:

```
module tool() {
    //simulate 1/8'' flat end mill
    cylinder(20,1.5875,1.5875, $fn=10);
}

/*
module tool() {
    //show toolpaths as thin lines or simulate 3d printer
    sphere(0.1, $fn=10);
}
*/

module path(a, b) {
    hull() {
        translate(a) tool();
        translate(b) tool();
    }
}

module gcode(vec) {
    for(i = [0:len(vec)]) path(vec[i-2],vec[i-1]);
}

//vector generated from g-code
vec = [
[0,0,0],
[0,0,-10],
[10,20,-10],
[20,10,-10],
[30,10,-10],
[30,10,0],
];

difference() {
    //stock material
    %translate([-10,-10,-20]) cube([50,40,20]);
    //object generated from g-code
    gcode(vec);
}
```
Output then represents the material that is added or subtracted from manufactured object:

[image]

Currently it does not read real g-code, it only takes list of coordinate vectors and moves the tool across them. Unfortunately it's currently not possible to interpret arcs and i don't know yet how to do that. it's simply not possible using hull(), because it always takes the shortest path between two points. I have to figure this out, because obviously it's not uncommon to have round shapes in g-code.

Anyway... Generating openscad model from g-code is not the hard part of CAM. It's the exact opossite: generating g-code from openscad model.

Harvie, 2017-05-01:
I think it might be possible to modify import() function in way it will be able to import g-code as 3d path. And then there can be some function that will follow that path with some object (pretty much the same stuff i am doing above). That would be workflow consisent with how you import 2D shape from DXF/SVG and then extrude it. It might be first easy step to have at least some quite usefull CAM functionality in OpenSCAD. Plus the 3D path follow subsystem can be usefull even for other stuff, not only for paths generated/imported from g-code. It's quite generic feature...

Neon22, 2017-05-02:
@Harvie possibly better to use one of the polygonsweep based approaches than Hull(). Faster and more control.

Currently no path or wire, or segment prims in openSCAD. I think it'd be useful and its been discussed before but not implemented. Closest thing we have is a list of tuples.

Once again I suggest you move this discussion to the forum.

rimmartin, 2017-05-28:
Hi
I don't see this discussion in the forum so asking here.

@Harvie from your experience you describe it as simple.  But you're experienced:-) Doesn't there need to be consideration of the stock material and material properties? Wasn't there a famous sculptor dude who said he removes what isn't the sculpture to get the desired object?

3d cad shows the end result but how does the g-code know to go from  stock material to it? What safe guards are there to not slam a cutter right into the bulk material? CNC machinists don't just take the g code file for granted?

Harvie, 2017-05-28:
@rimmartin in fact CNC machines are much dumber than you might think. as far as material is concerned, you need to set 2 things:

1.) you need to know dimensions of stock. especialy in Z direction, so you don't go milling to wise, or whatever holds the material in place. but this is usualy already kinda self-evident from model that you are going to use for generating g-code. also you check all dimensions in g-code sender before executing, so it's quite easy to be sure that you won't crash into anything.

2.) you need to set cutting parameters, which are usualy numbers manualy entered by CAM operator, which has experience and knows how to set that for combination of his machine, milling bit and material, or at least he looks it up in some table. This include stuff like RPM of spindle, feed rate (mm/min) for XY and Z (Z's usualy bit slover than X and Y), overlap and depth of cut. If you are CAD, all you need to do is pass this to CAM plugin and forget about it. Even CAM does not really care about speeds, because all CAM needs to do is to put that number on beginning of g-code, so machine knows how fast it should follow the generated path. OTOH the stuff like overlap and depth of cut has effect on CAD process itself, but it's not that hard either. Basically all it does is make each cut in several layers/passes with depth increments instead of cutting whole depth at once, which is not always possible. But this is not something that OpenSCAD would need to consider as there are already subsystems that will handle it. (eg. opencamlib). BTW FreeCAD recently got opencamlib support.

curiouspl2, 2018-04-22:
I think it's a great idea even if a little bit off the centre effort of openscad. 
Mainly because it provides 'crippled' CAM software decent UI and control, that is - allows calling external 
CAM processors. 

This allows easier sharing of whole CAD/CAM designs among amateurs - f.e. 3d printable object
could be equipped with a call to slicer right in openscad code, with all important parameters, including creation of f.e. modifier meshes, support structures, multi-material files, and additional commands f.e. for post-process painting or milling of part. 

While it sounds like big hassle and 'whole new product' it also allows using openscad in production environments, which hopes for more 'serious' (that is - paying money mortals have no access to ;) users into the game.

Harvie, 2018-04-22:
I think that first step should be importing g-code as 3D line to openscad. This shouldn't be that hard. It's not that different from importing DXF.

WillAdams, 2020-08-12:
As noted in #3400 simply having a series of modules which model the cutting from the bottom centerpoint would allow a user to export the G-Code commands for cutting to match the movement (it could be even better, since one could also write out G2/G3 arcs) --- all that's necessary is adding some commands to:

 - allow the user to define the output filename/extension
 - allow the user to add text to the file in question
 - add a menu command for writing out the file (or always write it out when processing it)

I've got a design I can't get cut for want of some straight line moves: http://forum.openscad.org/Managed-to-model-a-finger-joint-box-with-relieved-fingers-now-how-to-actually-cut-it-Allow-OpenSCAD--td29878.html and while I could put the G-Code in the Console and then parse that, that's a bit much to expect of a typical end-user.

Please just add this ability to do basic text output with full user control as a first pass (and generally useful option).

jordanbrown0, 2020-08-12:
It seems like the general problem that you're trying to solve is to somehow export metadata associated with various volumes in the resulting model.  That seems analogous to, for instance, generating multi-color models.

You might find that you want to model the volumes that are carved away rather than the volumes that remain.  It's the carved-away volumes where you need to know what tool to use, et cetera.

WillAdams, 2020-08-13:
If you would look at the code you would see that that is how I am working --- the problem is, while I was able to export multiple DXFs for the fingerjoints relieved with a V endmill https://community.carbide3d.com/t/design-into-3d-boxes-magazine-storage/16238/14, and also for square endmills: https://community.carbide3d.com/t/design-into-3d-rabbeted-box-with-features/20741/15 the radius endmill wants just a straight-line move and that's not something one can export using DXF or express using OpenSCAD.

What I want to do is to export specific coordinate data wrapped up in G-Code commands so that I can actually automate cutting out this design as opposed to drawing umpteen dozen straight lines.

jordanbrown0, 2020-08-13:
My point was mostly to point out the similarity (sort of) to multi-color models.

What I would kind of imagine would be to model a cut as a 3D construct - which it is, in real life - but include with it a hint that says "hey, CAM pipeline, this is a straight cut!".  Something like
```cam(type="cut") cube([length, kerf, depth]);```
would somehow emit a shape with, in addition to coordinates, the `type="cut"` annotation.

No, I don't have any idea how that would really work, especially if the object wasn't aligned with the XY plane on the stock, or wasn't something much like a cube with the kerf as the width.  My CNC experience is *very* limited.

Perhaps the `cam()` module would also accept parameters for things like the desired tool and the kerf.

And yes, this would imply a slicer-like component in the pipeline that would turn the geometry and hints into actual tool motions.

WillAdams, 2020-08-13:
Yes, that would be very cool for some analytical and automatic CAM module --- I'm not up to working on or thinking about that (and I find all of the existing tools which work thus frustrating and not suited to the sort of work that I want to do).

Given that I have defined a part definition in terms of tool movement, it would be nice to be able to capture the coordinates for said tool movement and write out a G-Code file which would then cut out the part --- doing that in OpenSCAD would allow doing the design all in one tool and would be simpler and more efficient.

As a proof of concept, I'll see about recreating this design in 2D in METAPOST and write out the G-Code from that.

nophead, 2020-08-13:
Why not use OpenSCAD to model the part in 3D and use a CAM program to
calculate the toolpath taking the tool geometry into account? That is what
the rest of the world does. It is only laser cutting where the line between
CAM and CAD gets blurred because the kerf is so small the tool path is
almost the same as 2D shape. And of course Fusion 3D where they have built
in a CAM backend.

WillAdams, 2020-08-13:
Because I can't find a CAM tool which will model cutting with a cove/corner/radius endmill, nor with other endmills with a sufficient accuracy for the sorts of joinery I'm trying to do --- please see the file: https://community.carbide3d.com/uploads/short-url/cb4ilRtY2Hestsi1kbgdht9yM0Z.zip esp. the module radiuscut.

Autodesk Fusion 360 isn't an option for me --- the new OpenSCAD feature I'm asking for is quite simple, just an extension of the existing console and would just involve one or two new commands and menu entries, then folks could do their own CAM in OpenSCAD if they wish (and ignore the new commands/menu entries if they don't want to trouble over them).

Harvie, 2020-08-13:
First of all. It is significantly more useful to have CAM integrated directly in CAD. Even FreeCAD now provides CAM. Both from user perspective and from the perspective of the fact that CAM can benefit from understanding the design of the part rather than just processing the 3D mesh. In such case the user have to kinda re-describe the operations that led to creating of the mesh. OpenSCAD already knows all the boolean operations and similar stuff, which i beleive can be leveraged to simplify the workflow.

Also currently there is no declarative CAM. i beleive having CAM integrated in OpenSCAD can significantly shift the way we think about CAM. Same way as OpenSCAD changed the way we think about CAD years ago. Providing us with means of doing lot of interresting stuff in automatic/programatic manner.

WillAdams, 2020-08-13:
Thank you.

Posted this to the mailing list:

Just adding a parallel output where if one called echo() only that text was written out to a second file (with a user configurable file extension) would work --- no added menu entry, just a pair of configuration options:

 [ ] write out echo text in separate file
 [ ] custom file extension, use ____ instead of .txt

that's all that would be necessary to allow folks to experiment with directly creating G-Code for CAM.

WillAdams, 2020-08-15:
Managed a proof of concept: https://community.carbide3d.com/uploads/short-url/Av8DyvPseArwbfWZcxKOLhoFQBW.zip

Parsing design (AST generation)... 
Saved backup file: C:/Users/willa/OneDrive/Documents/OpenSCAD/backups/Design into 3D_ Fingerjoint-backup-vaTVHyKD.scad

Compiling design (CSG Tree generation)...

ECHO: "G0 X,", 0, " Y", 0, " Z", 8
ECHO: "G1 Z", -3.175, " F", 150
ECHO: "G1 X", 0, " Y", 4.20625, " F", 600
ECHO: "G0 Z", 8
ECHO: "G0 X,", 9.5875, " Y", 0, " Z", 8
ECHO: "G1 Z", -3.175, " F", 150
ECHO: "G1 X", 9.5875, " Y", 4.20625, " F", 600

&c.

see the post at: https://community.carbide3d.com/t/cnc-finger-joint-box/8880/126

So if we just had a command which would write out only the echoed text less the clutter with a user-declarable file extension, we'd have a CAM solution for OpenSCAD files which are coded in a style where a declared piece of stock is subtracted from in a fashion which matches how one wants the CNC machine to cut things out.

WillAdams, 2020-08-20:
Wrote a post on this with a complete sample file at: 
https://community.carbide3d.com/t/doing-cam-in-openscad/25214 and mentioned it on the mailing list.

WillAdams, 2024-08-23:
FWIW, now that there is a working version of an "OpenPythonSCAD": https://pythonscad.org/
(I look forward to that being integrated into the main program as soon as is practicable)

I was able to work up a series of modules which afford the option of writing out G-code (as well as DXF) files: https://github.com/WillAdams/gcodepreview