openscad/openscad #2019: CMake build system

kintel, 2017-05-01:


thehans, 2017-05-03:
Regarding linux, I tried to run cmake ./ from a fresh openscad clone and got an error about my libxml version.

This is on Linux Mint 18.1 (based on Ubuntu 16.04)

[cmake.txt](https://github.com/openscad/openscad/files/972278/cmake.txt)

kintel, 2017-05-03:
@thehans I've downgraded the libxml2 requirement to 2.9.

FYI: The version requirements are currently not validated against our actual requirement, so feel free to report or fix such issues. Testing on Linux is very appreciated!

thehans, 2017-07-15:
Ok, the next errors I encountered were

```
CMake Error at CMakeLists.txt:381 (add_executable):
  Cannot find source file:

    src/MainWindow.cc

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
```
and then 
```
CMake Error at CMakeLists.txt:381 (add_executable):
Cannot find source file:

    src/AboutDialog.cc

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
```

After changing those to .h in CMakeLists.txt, cmake exits without error.  Log attached.
[cmake_log.txt](https://github.com/openscad/openscad/files/1149970/cmake_log.txt)

Then running `make`, I get this error during linking (full log also attached below):
```
[100%] Linking CXX executable OpenSCAD
/usr/bin/ld: CMakeFiles/OpenSCAD.dir/src/OffscreenContextGLX.cc.o: undefined reference to symbol 'XGetErrorText'
//usr/lib/x86_64-linux-gnu/libX11.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/OpenSCAD.dir/build.make:4493: recipe for target 'OpenSCAD' failed
make[2]: *** [OpenSCAD] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/OpenSCAD.dir/all' failed
make[1]: *** [CMakeFiles/OpenSCAD.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
[make_log.txt](https://github.com/openscad/openscad/files/1149971/make_log.txt)

So I think it needs `-lX11` somewhere?

thehans, 2017-07-16:
I think I got the Xlib included successfully, via #2069

After that change, the next error I see is some linking error related to QSettingsCached:
```
[100%] Linking CXX executable OpenSCAD
CMakeFiles/OpenSCAD.dir/src/mainwin.cc.o: In function `QSettingsCached::QSettingsCached()':
mainwin.cc:(.text._ZN15QSettingsCachedC2Ev[_ZN15QSettingsCachedC5Ev]+0x21): undefined reference to `QSettingsCached::qsettingsPointer'
mainwin.cc:(.text._ZN15QSettingsCachedC2Ev[_ZN15QSettingsCachedC5Ev]+0x42): undefined reference to `QSettingsCached::ctor_mutex'
mainwin.cc:(.text._ZN15QSettingsCachedC2Ev[_ZN15QSettingsCachedC5Ev]+0x54): undefined reference to `QSettingsCached::qsettingsPointer'
mainwin.cc:(.text._ZN15QSettingsCachedC2Ev[_ZN15QSettingsCachedC5Ev]+0x8a): undefined reference to `QSettingsCached::qsettingsPointer'
CMakeFiles/OpenSCAD.dir/src/mainwin.cc.o: In function `QSettingsCached::setValue(QString const&, QVariant const&)':
mainwin.cc:(.text._ZN15QSettingsCached8setValueERK7QStringRK8QVariant[_ZN15QSettingsCached8setValueERK7QStringRK8QVariant]+0x17): undefined reference to `QSettingsCached::qsettingsPointer'
CMakeFiles/OpenSCAD.dir/src/mainwin.cc.o: In function `QSettingsCached::value(QString const&, QVariant const&) const':
mainwin.cc:(.text._ZNK15QSettingsCached5valueERK7QStringRK8QVariant[_ZNK15QSettingsCached5valueERK7QStringRK8QVariant]+0x2a): undefined reference to `QSettingsCached::qsettingsPointer'
CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o: In function `QSettingsCached::remove(QString const&)':
Preferences.cc:(.text._ZN15QSettingsCached6removeERK7QString[_ZN15QSettingsCached6removeERK7QString]+0x13): undefined reference to `QSettingsCached::qsettingsPointer'
CMakeFiles/OpenSCAD.dir/src/Preferences.cc.o:Preferences.cc:(.text._ZN15QSettingsCached8containsERK7QString[_ZN15QSettingsCached8containsERK7QString]+0x13): more undefined references to `QSettingsCached::qsettingsPointer' follow
collect2: error: ld returned 1 exit status
CMakeFiles/OpenSCAD.dir/build.make:4520: recipe for target 'OpenSCAD' failed
make[2]: *** [OpenSCAD] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/OpenSCAD.dir/all' failed
make[1]: *** [CMakeFiles/OpenSCAD.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```
In case its not already painfully obvious, I don't actually know the first thing about cmake, but I think I'm slowly picking up some understanding of the CMakeLists.txt file.

thehans, 2017-07-16:
@kintel With the latest changes, the cmake build now completes without any apparent errors on my Linux machine.

thehans, 2017-07-17:
You're welcome.

I figured I might as well also post the logs from the build, so here they are with the commands used to create them:
`cmake ./ 2>&1 | tee cmake_log.txt`
[cmake_log.txt](https://github.com/openscad/openscad/files/1151440/cmake_log.txt)
`make 2>&1 | tee make_log.txt`
[make_log.txt](https://github.com/openscad/openscad/files/1151441/make_log.txt)

There were also a handful of warnings throughout the make, which I've clipped out from the full log, and pasted below:
```
AUTOGEN: warning: /home/hans/projects/openscad/src/grid.cc: file is empty
```

```
/home/hans/projects/openscad/src/MainWindow.ui: Warning: The name 'gridLayout_2' (QGridLayout) is already in use, defaulting to 'gridLayout_21'.
```

```
[  3%] [FLEX][openscad_lexer] Building scanner with flex 2.6.0
src/comment_parser.y: warning: 7 shift/reduce conflicts [-Wconflicts-sr]
```

```
In file included from /usr/local/include/CGAL/Triangulation_2_filtered_projection_traits_3.h:30:0,
                 from /home/hans/projects/openscad/src/cgalutils-tess.cc:10:
/usr/local/include/CGAL/internal/deprecation_warning.h:57:6: warning: #warning "A deprecated header has been included." [-Wcpp]
 #    warning "A deprecated header has been included."
      ^
/usr/local/include/CGAL/internal/deprecation_warning.h:58:54: note: #pragma message: Warning: The header <CGAL/Triangulation_2_filtered_projection_traits_3.h> is deprecated. Please use <CGAL/Triangulation_2_projection_traits_3.h> instead.
 #    pragma message (CGAL_INTERNAL_DEPRECATED_MESSAGE)
                                                      ^
```

thehans, 2017-10-31:
I've done more testing and I have good news to report.  Testing the cmake build on Linux Mint 18, with `ctest -C All`  I get 45 test failures, but they are the exact same test failure i get from the qmake build, so everything is consistent.

I did notice that the cmake build took about twice as long to complete tests, and tracked down the issue to the fact that qmake automatically add -O2 optimization, and cmake didn't.  (hence PR #2169 )

Here is a little bash script I wrote to test both builds, so you can see the commands I used.  It assumes dependencies are already met:
```
#!/bin/bash

git clone https://github.com/openscad/openscad.git openscad_cmake && cd openscad_cmake
git submodule update --init
cmake .
time make -j8
mkdir tests/build && cd tests/build
cmake ..
make -j8
time ctest -j8 -C All
cd ../../../

git clone https://github.com/openscad/openscad.git openscad_qmake && cd openscad_qmake
git submodule update --init
qmake CONFIG+=experimental
time make -j8
mkdir tests/build && cd tests/build
cmake ..
make -j8
time ctest -j8 -C All
cd ../../../
```

thehans, 2017-11-01:
I think it would be nice to do some testing of the cmake builds on travis.  Can we set up a branch (`cmake-build` since it already exists?) with an edited `.travis.yml` to invoke cmake instead of qmake and see how well its handled by travis?

kintel, 2017-11-02:
@thehans If you open a PR with such an edited travis.yml, it should automatically trigger Travis to do the right thing.

kintel, 2017-11-02:
You edited the "coverity" section, which is only run when we run the Coverity static analysis tests.
The main build is kicked off from `./scripts/travis-ci.sh`

thehans, 2017-11-02:
Ah, thanks @kintel.  Changing it in `travis-ci.sh` did cause travis to run cmake, so that's good.  
However the cmake command failed due to Issue #2082

thehans, 2017-11-02:
By the way, I ran the same ctest command as travis (copied from `scripts/travis-ci.sh`) on my Linux Mint 18 and 0 of 1064 tests Failed.

thehans, 2022-01-19:
cmake builds have fully replaced qmake now

closed 2022-01-19 by thehans