openscad/openscad #2020: Feature Tweak - Line marker numbering

MichaelAtOz, 2017-05-01:
I seem to frequently have the scale marker numbers not showing on the visible preview window.
For example from this:
[image]
One scroll wheel notch gives me this:
[image]

One possibility would be to have an intermediate stage where every 2nd mark is numbered.

DMCShep, 2017-08-18:
Screenshot 1 is at the closer zoom where every 10 units are labeled. Screenshot 2 is at the further zoom where every 100 units are labeled (you'd see the 100 labels if you panned a bit). Every ten marks shows a number. Took some screenshots of a guitar nut I just made demonstrating this.

Zoomed in to label every ten:
[image]

One click out to label every hundred:
[image]

Panning to show said label:
[image]

MichaelAtOz, 2017-08-19:
@DMCShep was that to just confirm the behaviour?

DMCShep, 2017-08-19:
Oops, thought the bug was the lack of any label, rather than it not being visible in the viewport.

In any case, detection of when the marks need to change scale should probably be independent of when the labels need to change scale. Wouldn't hurt to consider labels at every 10, 20, 50, 100, etc. units either.

MichaelAtOz, 2017-08-19:
This is a feature request. Could maybe a bug, but basically it was not helpful & I wanted to spend some bounty money on an annoying thing.

thehans, 2017-10-27:
I like the improved labeling scheme from @justbuchanan , but I have a suggestion to improve it further.

So the issue was mainly the size of the "jump" that the labels scale by was too large.

Originally it was a single jump * 10 for every decimal power.

So if are viewing a `cube(1);`  you start zoomed in enough to see single unit labels: 
1,2,3,4,5,...  then you start to zoom out ... and eventually the labels jump * 10 to =>
10,20,30,40,50,... and * 10 again, every time the label interval changes => 
100,200,300,400,500,...

With the latest changes, we now have effectively have two steps for each factor of 10.  first *2, then *5
1,2,3,4,5,...       * 2 => 
2,4,6,8,10,...      * 5 =>
10,20,30,40,50,...  repeat
So now the biggest jump we see is going from counting by twos to counting by tens: We've reduced that maximum jump in half from *10 to *5 

If we add in a third step of counting by fives (and maybe a corresponding medium tick size?) then that maximum jump can be divided in half again *2.5, which is more consistent with *2 jump of the other two steps now.
1,2,3,4,5,...  counting by ones     *2 => 
2,4,6,8,10,...counting by twos    *2.5 => 
5,10,15,20,25,... counting by fives *2 => 
10,20,30,40,50,... repeat

kintel, 2017-10-27:
I don't have a strong opinion - perhaps implement it (or a general mechanism for tuning it) and see if it feels better?

thehans, 2017-10-30:
I did a little more testing of the current code and found one issue in particular that I think should be addressed:
Zooming in beyond 0.2 label interval jumps to 0.02 (no "counting by ones" step), resulting in a lot of overlap and illegibility.  This gif shows a the difference between a single zoom step:
[image]

Also this is less severe, but I figured I'd mention the other worst case for the current labeling, which could potentially be alleviated with another labeling "step" as I mentioned in prev. post:
Zooming out to where label interval = 200000 can get crowded enough that labels flow together, and where the label interval beyond that is just out of frame.
[image]

I don't have a better implementation worked out yet, I could give it a shot, but I figured @justbuchanan gets first dibs if they want to take a crack at the problem. :)

justbuchanan, 2017-11-01:
@thehans, thanks for pointing that out, it's pretty unreadable. I just sent https://github.com/openscad/openscad/pull/2170 to fix this and it seems to work well. Please give it a shot and let me know if you see anything else.

justbuchanan, 2017-11-01:
Here's a quick test of adding labels every 5 tick marks: https://github.com/justbuchanan/openscad/tree/more-labels. You may be able to tune the thresholds better - I only adjusted them a little bit.

I think the overlapping labels for large numbers can also be fixed by adjusting the thresholds. There seems to be a bad zoom range when the numbers are large, but not large enough to switch to scientific notation. I'll play around with this some more.

Also, would there be any opposition to changing the font used for the numbers? I find the current one a little tough to read, especially because of the underscores used as decimal points.

thehans, 2017-11-01:
The challenge with changing the "font" is that all the characters are currently a bunch of hardcoded vertex data drawn as GL_LINES if I remember correctly.  Maybe it could be re-implemented to display a bitmapped font instead?

I think ideally the labeling and tick marks would be rendered directly by a fragment shader.

MichaelAtOz, 2017-11-01:
[KISS](https://en.wikipedia.org/wiki/KISS_principle)

kintel, 2017-11-01:
As Michael pointed out, this was optimized for developer time and to limit the number of dependencies.
One of the thing on the long-term TODO list is to rewrite the rendering infrastructure, perhaps into using some rendering/OpenGL framework. Once that's in place, such a framework may have support for font rendering, which would make this easier on us.

thehans, 2017-11-05:
Found another issue, the digits on the Z positive axis are in reverse order

[image]

justbuchanan, 2017-11-07:
I think you're right that the backwards numbering on the +z axis was that way before. Now that you've point it out, it looks obvious - I'm surprised I didn't notice it before.

I'm away from my dev computer right now, so I can't test this, but I suspect changing this line from  `di>0` to `di>=0` would fix it: https://github.com/openscad/openscad/blob/master/src/GLView.cc#L705. I can try it out later tonight or tomorrow.

justbuchanan, 2017-11-07:
I sent a fix for the backwards labeling for the z axis in #2186.

Also, +1 to your suggestion about the labeling on the y axis.

MichaelAtOz, 2018-02-19:
Is this fixed? @justbuchanan want to claim the bounty?

justbuchanan, 2018-03-25:
Hi @MichaelAtOz, sorry for the slow response, this message slipped past me. But yes, that would be great! I believe this is fully-implemented and working well.

MichaelAtOz, 2018-03-26:
Closing. @justbuchanan you can claim the bounty [here](https://www.bountysource.com/issues/44706972-feature-tweak-line-marker-numbering), let me know how it goes, as they have new owners.

closed 2018-03-26 by MichaelAtOz