openscad/openscad #2024: STL files missing from depencency file when output is PNG

nophead, 2017-05-03:
handle_dep() is called from import_stl(). When a png file is being generated the dependency file is written before the STL file gets imported, so the entry goes missing.

include and use call handle_dep() from the parser, so get added much earlier. import isn't handled by the parser as it is just a module name. I am guessing moving the handle_dep() call to ImportModule::instantiate would fix the problem.

kintel, 2017-05-04:
In general, handle_dep() should probably move out of the parser. We'd need to refactor include and use into propers AST nodes. That's on my radar, just hasn't happened yet.

kintel, 2017-05-04:
..which means that the dependency step could be one single AST traversal.

nophead, 2017-05-04:
The dependency information for include and use is also tracked in FileModule, which leads me to another bug. Changes to imported files don't trigger a reload.

nophead, 2017-05-04:
Is there a way to get hold of the parent FileModule from ImportModule::instantiate?

If there is I could add a new list of import dependencies to FileModule so that changes update correctly and I could also then remove all the handle_dep calls and just traverse the FileModule dependency tree to create the deps file.

nophead, 2019-02-24:
Fixed by #2025

closed 2019-02-24 by nophead