openscad/openscad #2026: Changes to imported files don't trigger auto compile.

nophead, 2017-05-04:
The imported files need to be added to the dependencies in the FileModule, so that changes trigger a recompile. The dependency file can then be created by walking the module tree rather than having a separate list of dependencies.

kintel, 2017-05-04:
Duplicate of #1232. See also #1400

closed 2017-05-04 by kintel

nophead, 2017-05-05:
Sorry for the duplicate. I can fix both of those if there is way to navigate from a module and a function to its parent FileModule.

kintel, 2017-05-06:
I don't think there is a direct way of finding the parent FileModule. The way to go may be to traverse the FileModule to discover the child resources.

nophead, 2017-05-06:
The trouble I see with that is they might not be static. For example of you have a module that imports a file named with a parameter then the dependency only becomes known when the module is instantiated and that is where handle_dep() is called in my current version.  I don't see how you can start from a FileModule and look for its import dependencies because they aren't necessarily visible from static code analysis, unlike includes and uses.

I am struggling to understand the context stuff.  When a function or module is instantiated it must have access to symbols in its parent FileModule by lexical scope. If it can get to those why can't there be a pointer to the FileModule that they belong to?

nophead, 2017-05-06:
I am thinking perhaps the getAbsolutePath() in Context should be replaced with a getFileModule() and FileModule gets getAbsolutePath() added. But I am confused about the comments in dxfdim about issue #217. Shouldn't EvalContext give access to the module where the function was instantiated?

nophead, 2017-05-06:
Actually scrub that. getAbsolutePath might be from an include file, not a FileModule.

kintel, 2017-05-06:
I agree that there are some pretty confusing bits in there. That's part of the reason why I started cleaning things up in a more bottom-up fashion; AST nodes, more general visitor system. There's still a bunch of work to do on the evaluator and context management side of things.

You're right that we cannot track dependencies from the AST - the node tree is probably a better place to do this. Since we do have general visitor infrastructure for the node tree, it should still be possible to write a dependency traverser which does only this.

nophead, 2017-05-06:
handle_dep() also does the auto make if the file doesn't exist, so it has to be run before the file is used. This is why it is in the lexer for includes and uses. 

I don't see how the visitor pattern helps in being able to get from the function and module instantiations back to the FileModule that owns them. Can a FileModule make a visitor visit all its instantiated members, match import module nodes and dxfdata function nodes and access their file path?

kintel, 2017-05-06:
I _think_ SurfaceNode and ImportNode do retain the full path to their resources without any need for consulting FileModule. Those could be picked up by traversing the node tree. These filenames are used during geometry evaluation.
For the dxf stuff it's a bit more annoying - I'm not to happy about the design of those. The filenames in question are known when calling the functions and these filenames are hence used during the AST evaluation.

Note: When `dxf_dim`/`dxf_cross` changes, we need to re-evaluate the modules they're used in (which is what FileModule does). If `import` or `surface` changes, we simply need to re-evaluate the geometry.

Not sure if this makes it easier. Is there anything more concrete I can do to simplify this?

nophead, 2017-05-07:
I was assuming this would fit into the existing update mechanism which reloads FileModules when their dependencies change. That would work if the dependencies could be reached from the FileModule, but as you point out, that isn't necessary for surface and import. However there isn't an existing mechanism that updates just geometry when a file changes is there?

Yes SurfaceNode and ImportNode store their filename. So a visitor could visit them all and see if the file has changed if an mtime is added to those LeafNode classes. If it does change what does the visitor do then? Is there an existing mechanism for invalidating / updating all the geometry that depends on a LeafNode that has changed?

Is it worth adding a completely new mechanism just to avoid recompiling a FileModule when an import changes?

dxf_dim and dxf_cross are just function pointers held in a BuiltinFunction object, so they can't be visited to get their filename as that is just a local variable in a function call. And they do need to cause their parent FileModule to update, so we need a way to associate the dependencies with the FileModule.  If this problem is solved then I think import and surface could simply use it rather than a whole new mechanism for incremental geometry updates. It is less than optimum performance wise but a lot better than not working at all.

nophead, 2017-05-07:
The lack of association between the instantiated nodes and FileModule / AST is also the reason why most of the error messages don't give the file and line number as would be expected.

kintel, 2017-05-07:
In terms of reacting to changes in SurfaceNode/ImportNode; I think we could just re-render. Those nodes retain the mtime of their resources as part of their cache key.

I agree that adding a way of associating every node with the AST could open a number of other doors (eventually we could be able to click on an object and see what source code created that object).

We might also keep in mind the numerous requests for extracting certain data from included files (like bounding boxes). If this will be supported in the future, we'd have to recompile anyway, so using the recompile mechanism may be the mose flexible approach.

A short-term fix could be to add a mechanism to FileModule for registering external file dependencies (for which we only check mtime). We could clear these dependencies for each compilation, and find an appropriate place for inserting them. A visitor is just an isolated way of encapsulating this logic, but there may be more natural places to put this code only for the two nodes in question. e.g. we already have a `FileContext` class which should serve as the parent context for evaluating any module inside file.  If we could retain a reference to the `FileModule` either in `ModuleContext` or `FileContext`, I think we should have enough information at the time of module instantiation to be able to register these dependencies. This can quickly get out of hand though : /

Btw., if you look into any of this, please have a glance at #1743, in case there is stuff there I've already mentioned or started on.

nophead, 2017-05-07:
Yes reading the code I was coming round to thinking it could be as simple as adding a reference to the FileModule in the FileContext.

I think we only check the mtime of the included file list so I think import files could be added to that list.