openscad/openscad #2027: Suggestion:  C-Like Vector Math

Corona688, 2017-05-10:
Since pretty much all data in OpenSCAD is vectors it's helpful to have lots of ways to use them.

In C you can do some basic vector-ops with plus.  "abcdefg" + 3 becomes "defg".  [1,2,3,4,5,6,7,8,9]+7 becomes [8,9].  [1,2]+19=crash, but in openscad, would just become undef.

This would simplify vector access that uses lots of coefficients.  a[x + (2*y)] could become (a+(2*y))[x] which may not look simpler at first, but remember that (a+(2*y)) is an expression you can pass into something.

oysteinkrog, 2017-05-12:
I'm not sure I understand 100% what you want.
[1,2,3,4,5,6,7,8,9] + 7 = [8,9] ??
How does that make any sense?

nophead, 2017-05-12:
It's C pointer maths, not vector maths, which OpenSCAD already does.

oysteinkrog, 2017-05-12:
Ah now I see.. I have to be honest, my gut instinct is horror!
Would array slicing not be good enough? (like python does it).

Corona688, 2017-05-12:
Array slicing would work but 
a) OpenSCAD already uses a similar syntax for loops, and it's better to not have the same syntax mean completely different things in slightly different contexts
b) It'd take changes to the language syntax, where this only requires installing an operator for the meaning of "vector + int".

doug-moen, 2017-05-12:
@Corona688 said "it's better to not have the same syntax mean completely different things in slightly different contexts"

I agree! That's why I think it's a bad idea for + to mean both addition and array slicing, in slightly different contexts.
```
[10,20] + [1,1] == [10,21]    // vector addition, as the language works today
[10,20] + 1     == [20]       // proposed syntax for array slicing
```
In just about every other language that supports vector addition, [10,20]+1 == [11,21], and array slicing is usually an extension of array indexing syntax. For example, we could define array[start:end] as an array slice.

oysteinkrog, 2017-05-13:
I agree very much with @doug-moen here.

MichaelAtOz, 2017-05-13:
& I agree with @oysteinkrog

kintel, 2017-05-15:
Closing this - let's try to avoid pointer arithmetics in OpenSCAD.

closed 2017-05-15 by kintel