openscad/openscad #2029: Avoid "blurry" surfaces with orthogonal view

JPFrancoia, 2017-05-11:
Hi,

I have an annoying phenomenon on OpenSCAD. As my english is not perfect, I'll simply show you 2 pictures. Top: before rendering, bottom, after rendering.

[image]
[image]

Can you see this brown-pixelated surface on the first picture ? Could you explain to me why it is there in the orthogonal view, and not in the perspective view ?

If possible, I would like to get rid of that. It is annoying, because I like the orthogonal view better.

Thanks !

tinkerology, 2017-05-11:
Extend your difference above the surface that you are removing the cylinder
from.  I define a value in all my models called DIFFERENCE_FUDGE=0.001 to
extend surfaces so that there are no coincident faces which cause that
issue.

Scott

t-paul, 2017-05-11:
See also the [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview.3F) for some more pointers. It's not ideal this happens, but it's an effect inherent to how the preview drawing works.

closed 2017-05-11 by t-paul

JPFrancoia, 2017-05-12:
This is clearly not ideal. I know I can use something like DIFFERENCE_FUDGE, but I don't want to. I don't want another variable in my design, they are already complicated enough. And when the design is complicated, using render() is also out of the question. Will it ever be fixed ?

nophead, 2017-05-12:
It won't be fixed as long as OpenCSG is used to generate a fast preview. It doesn't calculate geometry, it just draws pixels and relies on the graphics card's Z buffer to work out what is in front. When you have two coincident surfaces the depth of each pixel should be exactly the same but rounding to integer Z buffer values causes one surface to win in some pixels and the other at others.

JPFrancoia, 2017-05-12:
ah ok, now I get it. Ok, then I'll wait for a miraculous change someday...