openscad/openscad #2031: Can't view simple 3D objects in wire frame

MichaelAtOz, 2017-05-16:
I tried cube(20); & a cylinder, after rendering View/Wireframe just shows the same as View/Surfaces.

kintel, 2017-05-16:
We should clean up the menus. Wireframe only works for CGAL meshes, but it's not clear what makes a mesh a CGAL mesh. Forcing it through a boolean op should make it work.

MichaelAtOz, 2017-05-16:
You'd need to cleanup all the console output too; console for F6 cube(20)
```
Rendering Polygon Mesh using CGAL...
Geometries in cache: 137
Geometry cache size in bytes: 14054240
CGAL Polyhedrons in cache: 25
CGAL cache size in bytes: 204098640
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Facets:          6
Rendering finished.
```
What's the differences with a CGAL mesh v's whatever the other mesh is?

kintel, 2017-05-16:
We only create a CGAL mesh if necessary. If you've only got a single primitive, it's represented using a "PolySet" - a much simpler data structure. We can still do a wireframe rendering; it just needs to code to care for that special case.

PR #5530 (2025-03-07): Introduce PolySetRenderer and add support for wireframe overlay
Add support for wireframe overlay for Manifold-generated geometry:
* Separated out PolySet/Polygon2d-related rendering functionality from CGALRenderer to a new PolySetRenderer.
From now on, CGALRenderer is only used to render CGAL polyhedrons, and PolySetRenderer will be used whenever PolySets are generated (CGAL mode without CSG ops or Manifold mode).
* Rewrote the ViewEdges shaders to utilize color attributes instead of using uniform colors.

TODO:
* [x] Add tests

Fixes #2031
closed 2025-03-07 by kintel