openscad/openscad #2032: Feature Request: Command Line, Image export, option to include View features

MichaelAtOz, 2017-05-19:
I'm producing images using the png export in the GUI, and like to include the axes (which includes the three colour axis pointer, and the axis markers (dashes & numbers).
I need to make the image larger/higher resolution. On my new monitor I can get the GUI to ~1300 pixels high, but I'd like higher resolution to put the images on a T-shirt without pixelation.

I can do that with the command line, but I can't get the axes etc.

Of the View Menu options, I'd like Show Axis & Show Scale Markers.
That leaves Show Edges & Show Crosshairs, these should be added for consistency, but I'm in no hurry for them.
Whether anyone would want Wireframe as an option may need to be considered.

Current command line options re images are:
~~~
[ --viewall ] \
[ --imgsize=width,height ] 
[ --projection=(o)rtho|(p)ersp] \
[ --render | --preview[=throwntogether] ] \
~~~

If Wireframe is included, perhaps
~~~
[ --render[=(w)ireframe | --preview[=throwntogether] ] \
~~~

For the new stuff
`[--view=[(a)xis] [(s)cale] [(e)dges] [(c)crosshairs] ]`   <-- with commas, I just couldn't think of the right syntax

If doing that, perhaps consider adding `[(o)rtho] [(p)ersp] [all]` and deprecating the `--viewall` & `--projection` options??

MichaelAtOz, 2017-05-30:
I had a ligh-bulb-moment :bulb: changed my monitor to portrait.
[image]
I got a bit bigger image, but I should have got the full 4K monitor, I want 
:zap: Mooore :zap:  /play bezos ha ha ha ha cue-spooky-music /play horror /play bezos 
(/ play is the markup emoji sound effect for those reading this via email, don't know it it works here)

I need to adjust my workstation...my neck is killing me
[image]
(yes it has been a cold almost winter day, cabin fever...)

kintel, 2017-08-09:
I'll have a stab at the axis aspects of this.

MichaelAtOz, 2017-08-20:
Thanks, I'll test when a snapshot is available.

MichaelAtOz, 2018-02-19:
@kintel can this be merged into master so I can test via a snapshot. I'll do the doco then & local testing and see if I can help with the test suite tests. You can claim the bounty. 
(I'm going thru the bounty list as it's getting closer to the "No fees for 60 days!")

kintel, 2018-02-19:
@MichaelAtOz This is really only hinging on whether there are more options that should be added to `--view` before merging, but I'd be happy to merge early and leave that as a future task.

MichaelAtOz, 2018-02-19:
@kintel Merge, close, claim. I'll raise another issue to cover tidying up the other command line options as it isn't time sensitive & low-hanging for someone. 
Now I have my hand back I'll do some more T-shirt stuff, the new `--view` will be immediately useful.

t-paul, 2018-11-01:
CI-Builds:
32-bit: https://circleci.com/gh/openscad/openscad/275#artifacts/containers/0
64-bit: https://circleci.com/gh/openscad/openscad/276#artifacts/containers/0

Those should be accessible without login... for some reason the "Artifacts" tab does not show up but it seems to still work via the direct link.
Maybe we can find a way to put that on the download page (as every new branch will have windows builds :smile:)

closed 2018-11-04 by kintel

MichaelAtOz, 2018-11-04:
Thanks. Anyone want to claim the bounty?

commit 6e6e38d8 (2018-11-02): Add view options for command line (fixes #2032).
commit ec8dd595 (2018-11-04): Add view options for command line (fixes #2032).