openscad/openscad #2034: Union of manifold STL files produces non-manifold output

Jip-Hop, 2017-05-19:
I've batch exported several cubes from Blender to STL and unioned them in OpenSCAD.
Each STL contains just a cube and is manifold.
OpenSCAD doesn't give any errors when rendering and exporting to STL.
However the result of the union is non-manifold according to blender and Meshlab.
```
union(){
    import("0019.stl");
    import("0018.stl");
    import("0017.stl");
    import("0016.stl");
    import("0015.stl");
    import("0014.stl");
    import("0013.stl");
    import("0012.stl");
    import("0011.stl");
    import("0010.stl");
    import("0009.stl");
    import("0008.stl");
    import("0007.stl");
    import("0006.stl");
    import("0005.stl");
    import("0004.stl");
    import("0003.stl");
    import("0002.stl");
    import("0001.stl");
}
```
I realize I could generate the cubes directly in OpenSCAD but I'm investigating a workflow to generate models suitable for 3D printing and ran into this issue. Reason I'm using OpenSCAD is union, hollowing with wall thickness and rounding edges. This doesn't work reliably in Blender. But what I miss from Blender is the ease of interactivity and dragging some objects around. Therefore I'm interested in OpenSCAD <-> Blender round-tripping.

[non-manifold-output.zip](https://github.com/openscad/openscad/files/1013919/non-manifold-output.zip)

nophead, 2017-05-19:
A CSG union of manifolds does not necessarily produce a manifold. For example simply unioning two cubes with a shared edge produces a non-manifold result.

Jip-Hop, 2017-05-19:
I wasn't aware of this restriction. I thought such issues would only arise with difference or intersection, not union. But now that I look at your example I get it.

I'm extruding 2D shapes and then OpenSCAD doesn't give a warning. These two pieces of code produce the same shape.
```
linear_extrude(height = 1){
    union(){
        square([1,1]);
        translate([1,1,0]){
            square([1,1]);
        }
    }
}
```
```
union(){
    cube([1,1,1]);
    translate([1,1,0]){
        cube([1,1,1]);
    }
}
```
But only in the second case OpenSCAD warns me about this issue: `WARNING: Object may not be a valid 2-manifold and may need repair! `.

Can you mention all the exceptional cases that could prevent manifold output when using the union modifier? I'm making 3D models consisting of just stacked cubes, based on some data. Cases like you mentioned, two cubes with a shared edge, will be present at times. I didn't program any checks to work around this limitation because I didn't realize it would cause issues.

[image]

I've been having issues with this in Blender for a while ([posted on Stack Exchange](https://blender.stackexchange.com/questions/79023/union-partially-overlapping-cubes-without-boolean-modifier)) and thought I solved it by switching to OpenSCAD. But seems I'm going to have the same issues.

So far I've detected these cases:

- (partially) shared edge
[image]
[image]

- shared vertex
[image]

It's still a bit strange, because to me this looks perfectly 3D printable although it isn't a manifold mesh:
[image]

RonaldoCMP, 2017-05-19:
I gess that the union of two cubes is a non-manifold only if their
intersection is a point or a straight line segment.

nophead, 2017-05-20:
Also if they share part of a face. They must overlap with a finite volume, or not overlap at all. If they only share a vertex, edge or face the overlap is not a 3D shape and has no volume.

Jip-Hop, 2017-05-20:
Thanks for the lesson. Clearly this isn't an issue with OpenSCAD then. In my program I now check for these cases and increase the scale by a tiny amount so the cubes actually intersect instead of just touch. I think it's working now.

closed 2017-05-22 by Jip-Hop