openscad/openscad #2037: Enforce batch parameter -D variable=value  when using a customizer data set

PRouzeau, 2017-05-22:
When selecting a customizer dataset in batch mode, using a parameter -D variable=value won't supersede the value of the customizer dataset. As generally dataset are recorded and contains all variables, you cannot by default impose a variable in batch mode. You shall manually remove any variable you want to modify from the dataset, which is tedious.
I think -D parameter shall have priority over the value defined in the dataset. 
See an example of trouble caused by the present behaviour in my comments in issue #1999

maave0, 2026-10-11:
Bumping this so it may be considered. I have a build script that is supposed to pass in certain sizes

```
openscad -o build/"${filename}" -D "case_type_override=\"$case_type\"; case_material_override=\"$case_material\"; case_thickness_override=\"$case_thickness\"; version=\"$version-$git_commit\";" -p phone_case.json -P "${model}" phone_case.scad
```

but customizer parameters override `-D` params so I have to use a ternary and a second variable
```
 /* if a variable is defined in JSON, I can't override it by command line */
case_type_override="stupid_hack";
case_type2 = (case_type_override!=undef && case_type_override!="stupid_hack") ? case_type_override : case_type;
```

jordanbrown0, 2026-10-11:
We should either plan to change this, or explicitly decide that this is the way it is.

Of course, one workaround is to *not* specify a customizer data set when using the CLI.

PRouzeau, 2026-10-11:
The workaround I use is to create an intermediate variable which will be used by the program. This intermediate variable can be modified by the command line. This is simple and can be easily be systematized.
<pre>
//This is parameter1
cvar1 = value1;
var1 = cvar1;
//This is parameter 2
cvar2 = value2;
var2 = cvar2;
</pre>

cvar1 and cvar2 are set by the dataset. 
var1 and var2 are used by the program and they are modifiable by the command line.

I did explain this method  in the paragraph 'Using datasets' of the chapter 'Automation' I wrote on the manual.
[https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment#Automation](url)