openscad/openscad #2039: Segmentation Fault on closing

amarjeetkapoor1, 2017-05-25:
I am getting a segmentation fault on closing openscad (recent build). 

How to replicate:
Just run Openscad and open any file and then simply close.

```
amarjeet@amarjeet-Vostro-3559:~/mountPoint/projects/openscad$ ./openscad
Segmentation fault (core dumped)
```

kintel, 2017-05-26:
Random idea: Did you make sure to do a `make clean` and rerun qmake?

amarjeetkapoor1, 2017-05-26:
Yes, I did make clean and run qmake. I just Debugged and the problem was related to QSetting and it is caused after commit https://github.com/openscad/openscad/commit/25cde91df5e35d5f92a79c73fb209e41b246af13

amarjeetkapoor1, 2017-05-26:
Plus also getting "Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged." when any dialog box is opened like save or open.

DrCodeWizzard, 2017-07-26:
Since I made that commit... Did your debugging session give hints about what exactly is causing the crash dump?

amarjeetkapoor1, 2017-07-29:
I have already shared what I was able to get after debugging.