openscad/openscad #2040: difference doesn't work

Kirchner69, 2017-05-25:
It's my first project with OPENscad and now I have a Problem that I don't understand and I think it must be a bug.

Win 7 32 and 64 Bit 
Openscad version 2015.03-2 

Problem: I want to make a Difference of the hole object and the silver cube at the end of the code, but it doesn't work proper.
Picture 1: bevor difference
[image]
Picture 2: after difference
[image]

Code: 
`$fn=100;
Stockdurchmesser=20;

Staerke=2;
Laenge=150;
Abstand=100;

// Problem with differende over all Objects
//differende()}

// Stockfassung
color("red")
translate([0,0,Abstand-11])
difference(){
cylinder(25, Stockdurchmesser/2+Staerke,Stockdurchmesser/2+Staerke); 

translate([0,0,-1]) 
cylinder(32, Stockdurchmesser/2,Stockdurchmesser/2); 
}

difference(){
// Stockbefestigung
rotate([0,0,45])
translate([0,0,Abstand+6.5])
difference(){
translate([0,-10,0])
color("Blue")
cube([Stockdurchmesser-6,(Stockdurchmesser/2+Staerke+10+3),15], center=true);
    
// Loch von oben    
translate([0,0,-1]) 
cylinder(32, Stockdurchmesser/2,Stockdurchmesser/2, center=true);    
     
// Schraubenlöcher
translate([0,-(Stockdurchmesser/2+5+3),0])
rotate(a=[0,90,0])
cylinder(50,3,3, center=true);

// Loch von oben durch Schraubenbefestigung
cylinder(32, Stockdurchmesser/2,Stockdurchmesser/2);
   
}
}

// Halbkugel
difference(){
sphere(Abstand);
sphere(Abstand-10);

mirror([0,0,1])
cylinder(Abstand, Abstand, Abstand);

//Loch für Stock
translate([0,0,Abstand/2-5]) 
cylinder(Abstand+50, Stockdurchmesser/2,Stockdurchmesser/2, center=true);

// Halbkugel ausschneiden
translate([Abstand/2+5, -(Abstand/2+5),Abstand/2-7.5])
cube([Abstand,Abstand,Abstand+20], center=true);
translate([-(Abstand/2+5), -(Abstand/2+5),Abstand/2-7.5])
cube([Abstand,Abstand,Abstand+20], center=true);
translate([-(Abstand/2+5), Abstand/2+5,Abstand/2-7.5])
cube([Abstand,Abstand,Abstand+20], center=true);
translate([Abstand/2+5, Abstand/2+5,Abstand/2-7.5])
cube([Abstand,Abstand,Abstand+20], center=true);
}

// Problem with difference
rotate([0,0,45])
translate([0,-15,Abstand+6.5])
color("silver")
cube([Stockdurchmesser-Staerke*2-6,(Stockdurchmesser/2+Staerke+10+3+2),20], center=true);
//}
`

jmacarthur, 2017-05-25:
In your example you have four objects in your commented-out difference - so it is subtracting everything from the red element.

If you want to subtract just the silver object from the others, then you should put the first three objects (from 'red' to just before the `//Problem with difference` comment, like this:

```
// Problem with differende over all Objects
difference(){
union(){
  // Stockfassung
  color("red")
  translate([0,0,Abstand-11])
.
.
.
    cube([Abstand,Abstand,Abstand+20], center=true);
    }
  } // union ends here
  // Problem with difference
  rotate([0,0,45])
  translate([0,-15,Abstand+6.5])
  color("silver")
  cube([Stockdurchmesser-Staerke*2-6,(  Stockdurchmesser/2+Staerke+10+3+2),20], center=true);
}
```

That should subtract the silver object from the union.

Kirchner69, 2017-05-26:
So i have learned a important new think fpr OPENscad.
Thank you very much!

closed 2017-08-04 by kintel