openscad/openscad #2041: upgraded openscad from the last 2014 64 bit version to 2015  64 and tried 32bir versions

T3DT, 2017-05-25:
All 2015 versions of Openscad crash at startup 
On my Windows 7 Home Premium Service pack1 Intel i3 M350 laptop 8gb ram

So to re-cap 
OpenSCAD-2014.03-x86-64 Works ok
OpenSCAD-2014.03-x86-32 Works ok

OpenSCAD-2015.03-1-x86-64 Crash
OpenSCAD-2015.03-1-x86-32 Crash

OpenSCAD-2015.03-2-x86-64 Crash
OpenSCAD-2015.03-2-x86-32 Crash

T3DT, 2017-05-25:
I uninstalled OpenSCAD-2014.03-x86-64 Works then installed OpenSCAD-2015.03-2-x86-64

I uninstalled OpenSCAD-2015.03-2-x86-64 then installed OpenSCAD-2015.03-1-x86-64

I uninstalled OpenSCAD-2015.03-2-x86-64 then installed OpenSCAD-20145.03-x86-64

I performed an upgrade process by installing the new versions over the old version. 

I repeated the above for the 32bit versions with the same results.. 

I see enough of the 2015 version to see what looks like a UI redesign or major change in the look

T3DT, 2017-05-25:
Any suggestions of a possible workaround or other things to test would be appreciated

As the build at fault is now two years old..

I also tested the latest builds i.e. the 2017 Development Snapshots exactly the same result.

So I asume it is connected with the UI changes made as 32 and 64 bit versions are the same

nophead, 2017-05-25:
I would suspect your graphics driver and look to see if there are any
updates.

kintel, 2017-05-26:
Could you open the Windows Event Viewer and see if there are any messages there related to the crash? That tends to give us a hint about which component caused the crash.

MichaelAtOz, 2017-05-26:
While testing different versions you can also download the zip file, then extract to 'program files' 64bit or 'program files(x86)' 32bit.
Run the exe from there.
Easier than doing repeated installs or uninstalls.
Once you get it working, then you could install.

T3DT, 2017-05-27:
NopHead 
Intel(R) HD Graphics  Windows updater reports an up to date driver 20-02-10  

kintel 
Here is the Event Viewer Report

Log Name:      Application
Source:        Application Error
Date:          27/05/2017 01:17:33
Event ID:      1000
Task Category: (100)
Level:         Error
Keywords:      Classic
User:          N/A
Computer:      BodgeItQuick_lp
Description:
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x56481b58
Faulting module name: ig4icd64.dll, version: 8.15.10.2086, time stamp: 0x4b8006e0
Exception code: 0xc0000005
Fault offset: 0x000000000068bfb6
Faulting process id: 0x234c
Faulting application start time: 0x01d2d67e9eb0e230
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: C:\Windows\system32\ig4icd64.dll
Report Id: e052ba8f-4271-11e7-bb71-705ab6f26e3b
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2017-05-27T00:17:33.000000000Z" />
    <EventRecordID>157158</EventRecordID>
    <Channel>Application</Channel>
    <Computer>BodgeItQuick_lp</Computer>
    <Security />
  </System>
  <EventData>
    <Data>openscad.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>56481b58</Data>
    <Data>ig4icd64.dll</Data>
    <Data>8.15.10.2086</Data>
    <Data>4b8006e0</Data>
    <Data>c0000005</Data>
    <Data>000000000068bfb6</Data>
    <Data>234c</Data>
    <Data>01d2d67e9eb0e230</Data>
    <Data>C:\Program Files\OpenSCAD\openscad.exe</Data>
    <Data>C:\Windows\system32\ig4icd64.dll</Data>
    <Data>e052ba8f-4271-11e7-bb71-705ab6f26e3b</Data>
  </EventData>
</Event>

kintel, 2017-05-27:
@T3DT It crashes in `ig4icd64.dll` which is "OpenGL(R) Driver for Intel(R) Graphics Accelerator". A driver shouldn't crash, so I'd definitely start by upgrading the OpenGL drivers. This driver version, 8.15.10.2086, is at least 7 years old so I'm sure Intel has made some progress fixing bugs since then :)

T3DT, 2017-05-27:
Found this link for ig4icd64.dll I put the new file in the root of Openscad this has stopped the crash

But fails to change the display area of the rendered object all that is seen where the rendering should be is whatever there is behind the openscad window.

[image]

https://www.opendll.com/index.php?file-download=ig4icd64.dll&arch=64bit&version=9.17.10.2875&dsc=OpenGL(R)%20Driver%20for%20Intel(R)%20Graphics%20Accelerator#

kintel, 2017-05-27:
You probably have to get a proper installation package for this driver, as the DLL alone is unlikely to fix it.

kintel, 2024-03-27:
Closing this as a suspected driver issue

closed 2024-03-27 by kintel