openscad/openscad #2043: Add option to export OpenSCAD model to PNG heightmap

Harvie, 2017-05-27:
Hello,
currently this is possible:

```
scale([1, 1, 0.1])
  surface(file = "smiley.png", center = true);
```

transforming this

[image]

into this

[image]

I think it would be cool to have exact oposite... taking OpenSCAD model and exporting it to grayscale PNG heightmap. It does not have to be overly complex. Always taking Z axis as height, while 0 and lower is always black can simplify everything a bit.

atlaste, 2020-02-08:
Same here, that would be great for cnc rendering. Is that possible?

jongra, 2021-06-14:
Has anyone already accomplished this in some way?

maecki-maecki, 2024-01-04:
Using dmap2gcode this would be an easy way to fabricate komplex pieces with 2-sided cnc milling.

Mäcki