openscad/openscad #2045: Feature request: Add filename to console for export

MichaelAtOz, 2017-05-29:
ATM the console shows e.g.
```
...
Rendering finished.
STL export finished.
CSG export finished.
etc.
```
For Shapeways, different materials have different Design Guidelines, such as Minimum Wall Thickness, Clearance etc.
Thus often I need to export multiple STLs, with a change to a relevant variable to suit a particular material.
I do this by including something as a postfix to the filename.
This can get confusing. I often switch windows while the STL is produced.
So knowing the last filename will inform of where I was up to.

So something like `STL export finished. some_file_name.stl`
or, to preserve the existing message, in case some poeple process it.
```
Exporting some_file_name.stl
STL export finished.
```

A possible related issue is whether the console should be cleared at each compile/preview/render.
I think it would be good to be able to scroll back beyond the last operation.
This probably needs discussion??

t-paul, 2019-03-16:
Looks like this is already implemented. Only exception was CSG which is now covered by #2872.

closed 2019-03-16 by t-paul

MichaelAtOz, 2019-03-16:
> Only exception was CSG which is now covered by #2872.

And Export/Image, which I [mentioned](https://github.com/openscad/openscad/issues/2845#issuecomment-470369146).

t-paul, 2019-03-16:
Hmm, I skipped that as it's not a normal export and did not even have any info, but we can add that too...

t-paul, 2019-03-16:
#2875