openscad/openscad #2047: Openscad ubuntu16.04 does not render properly anymore

usernamereid, 2017-06-01:
Something strange is going on...
Iḿ running an old Dell D530 with 32bit ubuntu 16.04 LTS and openscad 2015.3-01

I recently had an update from ubuntu and then openscad stopped rendering properly.

see attachments, before and after, with the same openscad file.

[image]
[image]
 and the resulting stl file looks even worse now  - it looks like the $fn function is messed up a bit.
I also attach the code.scad file and a piccie of the STL file in my flashforge finder software.

```
$fn=100;
difference(){
cylinder(h=0.5,r=16);
translate([0,0,0])
cylinder(h=1,r=11);
}
difference(){
    color("grey")
cylinder(h=55, r=12.5, center=false);
translate([0,0,-.5])
cylinder(h=65, r=11.5, center=false);
translate([0,25,20])
rotate([90,0,0])
cylinder(h=50, r=2,center=false);
translate([0,3,20])
rotate([90,0,0])
cylinder(h=50, r=4, $fn=6, center=false);
}
difference(){
translate([0,13.5,20])
rotate([90,0,0])
cylinder(h=3, r=4.8, center=false);
translate([0,25,20])
rotate([90,0,0])
cylinder(h=50, r=2,center=false);
}
difference(){
translate([00,-11.3,20])
rotate([90,0,0])
cylinder(h=2, r=4.8, center=false);
translate([0,2,20])
rotate([90,0,0])
cylinder(h=50, r=4.1, $fn=6, center=false);
}
difference(){
translate([0,0,44])
sphere(20, $fn=100);
sphere(19, $fn=100);
translate([0,0,-.5])
cylinder(h=55, r=11.5, center=false);
 }
//support
 offset=17;
 translate([0,0,offset])
 difference(){
 supportring(offset);
 }
  module supportring(){
 difference(){
color("blue")
cylinder(h=12.0, r=12.4,center=false);
translate([0,0,-0.5]) 
color("pink")
cylinder(h=13.0, r=9.0, center=false);
translate([0,0,20-offset])
rotate([90,0,0])
color("green")
cylinder(h=100, r=1.5,center=false);
// recessed nut
translate([0,0-10,20-offset])
rotate([90,0,0])
cylinder(h=3, r=4.1, $fn=6, center=false);
 color("cyan")
translate([0,25,20-offset])
rotate([90,0,0])
cylinder(h=50, r=2, center=false);
 }
}
![screenshot from 2017-05-31 22-33-24](https://cloud.githubusercontent.com/assets/29106270/26696056/673dbfea-470d-11e7-8b88-8e4fad57d790.png)
![screenshot from 2017-05-31 22-32-59](https://cloud.githubusercontent.com/assets/29106270/26696067/71d69968-470d-11e7-9102-53fd90d9d240.png)

any ideas???
thanks
David

```

usernamereid, 2017-06-01:
Additional info after playing  around tonight.
If I remove the color statements with // I can import the model into openjscad and it looks perfect there, then exporting as STL in ASCII mode fails when I load the STL into my flashforge software, but saving the STL as binary, loads into Flashforge with only a complaint that the model needs fixing. After fixing it renders perfectly. and I can print it as it should be...see piccie.
[image]

t-paul, 2017-06-01:
The model looks totally fine. You have "Edge mode" enabled which looks a bit strange due to the dense vertices. Click the last icon below the view port or uncheck "View -> Show Edges".

usernamereid, 2017-06-01:
Hi Paul,
thanks for the tip... now the preview looks good, the rendering via F6 also looks good... but...when I import the created STL file into flashforge.. I get:
[image]

t-paul, 2017-06-01:
I'd suggest loading in a different program to check (e.g. meshlab). I can't see anything wrong with the exported STL here, so it might be an issue with that FlashForge software.

Failing all that, it would be interesting to look at the actual STL file generated on your system.

usernamereid, 2017-06-01:
I will post it tomorrow as it is after midnight here in the netherlands...

usernamereid, 2017-06-02:
Hi Paul,
as promised, the STL file that fails...I'm interested to hear if it renders OK for you.
(or if there is something wrong in the STL.... I have printed this file before, without and error when loading the STL into flashforge....

[light7x.stl.zip](https://github.com/openscad/openscad/files/1048591/light7x.stl.zip)

t-paul, 2017-06-02:
That looks fine in both meshlab and blender. If that's the same file as shown in the screenshots above loaded into that FlashForge software, I'd say it's a bug there.

usernamereid, 2017-06-02:
Hi Paul,
Ok great to know I will contact them and see if they can sort it. Really appreciate your help and knowledge!
David

closed 2017-06-02 by usernamereid