openscad/openscad #2053: Allow let() with tail recursion

arpruss, 2017-06-07:
It would be nice if one could use let() with tail recursion optimization. Here are three patterns that should work for tail-call optimization:
    function f(x,y,z) = let(...) ( condition ? returnValue : f(...) );
    function f(x,y,z) = ( condition ? returnValue : let(...) f(...) );
    function f(x,y,z) = let(...) ( condition ? returnValue : let(...) f(...) );

Currently, tail-call optimization works with let() when the let() is within the function arguments. But allowing the let(...) to be outside the function would allow one to avoid re-computing things in some contexts. For instance, one could then do `let(a=expensiveFunction(x,y,z)) f(a+1, a+2, z-1)` as the tail instead of `f(expensiveFunction(x,y,z)+1,expensiveFunction(x,y,z)+2,z-1)`.

thehans, 2017-06-22:
I am running into some recursion issues with my latest project, and created a small test case to better understand what was going on.

```
function countdn(n, result=[] ) = let (r = concat(result, [n])) (n > 1) ? countdn(n-1, r) : r;
function countup(n, result=[] ) = let (r = concat([n], result)) (n > 1) ? countup(n-1, r) : r;
function countdn2(n, result=[]) = (n > 0) ? countdn2(n-1, concat(result, [n])) : result;
function countup2(n, result=[]) = (n > 0) ? countup2(n-1, concat([n], result)) : result;

module test_all(n) {
  echo("dn2", len(countdn2(n)));
  echo("up2", len(countup2(n)));
  echo("dn", len(countdn(n)));
  echo("up", len(countup(n)));
}

test_all(6000);
test_all(8000);
```
On my computer this gives the output:
ECHO: "dn2", 6000
ECHO: "up2", 6000
ECHO: "dn", 6000
ECHO: "up", 6000
ECHO: "dn2", 8000
ECHO: "up2", 8000
ERROR: Recursion detected calling function 'concat'

RonaldoCMP, 2017-06-25:
See
https://en.m.wikibooks.org/w/index.php?title=OpenSCAD_User_Manual/User-Defined_Functions_and_Modules&action=edit&section=3

RonaldoCMP, 2017-06-26:
The link I referenced is for edition. A better link would be:
 https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Recursive_functions
<https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/User-Defined_Functions_and_Modules#Recursive_functions>

2017-06-24 21:02 GMT-03:00 Ronaldo Persiano <rcmpersiano@gmail.com>:

> See https://en.m.wikibooks.org/w/index.php?title=
> OpenSCAD_User_Manual/User-Defined_Functions_and_Modules&
> action=edit&section=3
>

thehans, 2017-06-26:
@RonaldoCMP I've read that before, what is your point?

RonaldoCMP, 2017-06-26:
Your first two recursive functions are not entitled to tail recursion
elimination and are limited to a lower recursion level due to stack
overflow.

2017-06-26 18:36 GMT-03:00 Hans L <notifications@github.com>:

> @RonaldoCMP <https://github.com/ronaldocmp> I've read that before, what
> is your point?
>

arpruss, 2017-06-26:
Yes, indeed. But the first two recursive functions are equivalent to syntax that in some other languages is friendly to tail recursion elimination. That's the point of my suggestion: extend tail recursion elimination to allow this kind of use of let().

jschobben, 2019-07-29:
This should be fixed once PR #3020 passes the test of scrutiny.

t-paul, 2019-09-15:
Closing as #3020 is merged now.

closed 2019-09-15 by t-paul