openscad/openscad #2054: openscad sleeps on OSX 10.12.5

akebono3, 2017-06-08:
App Nap in OSX applies to openscad, so the app thread sleeps while in the background.  For the user, the only option in 10.12.x is to issue 
`defaults write NSGlobalDomain NSAppSleepDisabled -bool YES`
which does not allow any app to sleep.

You can verify the apps status in Activity Monitor / Energy.

akebono3, 2017-06-08:
Update: according to for a single app at [apple.stackexchange](https://apple.stackexchange.com/questions/121386/missing-prevent-app-nap-button-on-app) in Mavericks, the no-sleep command is

> You can disable App Nap for a particular Application by going to Terminal.app and typing:
> `defaults write <app domain name> NSAppSleepDisabled -bool YES`
> Being the `<app domain name>` the application's name and its company's, with the following format: `com.companyname.appname`

It isn't clear that this would work in Sierra.

kintel, 2017-08-04:
Any idea how this should be fixed?
Note: We render in a separate thread, so having the UI thread sleep shouldn't affect us, unless App Nap sleeps the entire process.

akebono3, 2017-08-04:
Unfortunately, I'm not an expert in App Nap or in compiling on the OSX platform.  I'm just a user.  What I do know is that the render thread goes to sleep since the CPU goes idle after the OpenSCAD UI is put in the background with an active render.  It will wake up when the OpenSCAD UI is placed back in the foreground.

The documentation I linked to earlier indicates that OSX uses "advanced heuristics" to determine whether to idle a thread.  It seems possible that all that may be required is a flag.  Alternatively, the stackexchange discussion on OSX 10.9 indicates that all that needs to be added is a kMdItemCFBundleIdentifier needs to be created so that the thread can be identified and prevented from sleeping.  I have no idea if that feature is deprecated in OSX 10.11 or 10.12 .  

It could be dead simple, or require significant reading of developer documentation.  Maybe this is a clue: [https://forum.libcinder.org/topic/os-x-how-to-keep-rendering-in-background ]
Specifically, 

> To support activities that should not be suspended, the NSProcessInfo class has three new methods—beginActivityWithOptions:reason:, endActivity:, and performActivityWithOptions:reason:block:—to tell OS X that your app is actively doing something important.