openscad/openscad #2055: Grammar: "The Programmers Solid 3D CAD Modeller" is missing an apostrophe

deisner, 2017-06-08:
In the tagline (on [www.openscad.org](http://www.openscad.org/), for example), s/Programmers/Programmer's/.

MichaelAtOz, 2017-06-08:
Well, I think we should trash that slogan. Get a snappier millennial hipster maker one...

nophead, 2017-06-08:
Yes they are all Coders now, although I hate that term.

thehans, 2017-07-02:
If we fixed the grammar, then we'd have to call it "The Pro-grammar's Solid 3D CAD Modeller" though.
:stuck_out_tongue:

MichaelAtOz, 2017-07-02:
But every one nows programmers are betterer at programming than grammer, biggly...

DavidEGrayson, 2019-06-26:
That prominent typo is embarrassing.  Does it properly represent the quality of the rest of the project?

kintel, 2019-06-26:
It does probably represent that most of our developers' native language is closer to C++ than English

MichaelAtOz, 2019-06-27:
Isn't that the plural, the 'solid 3D CAD Modeller' for all the programmers?

As the Merch is about to be on top of my list can we lock this down.

Google 'programmers', we are not alone.

[Also](https://www.merriam-webster.com/words-at-play/history-and-use-of-the-apostrophe)
> Some like to think of the English language as an orderly sort of thing, a rock of stability in the chaos and uncertainty of life. Yet this is far from true: our language is a sloppy mess. Consider the apostrophe.
The 's' at the end of a word indicating possession ("The king's fashion sense") probably comes from the Old English custom of adding '-es' to singular genitive masculine nouns (in modern English, "The kinges fashion sense"). In this theory, the apostrophe stands in for the missing 'e'.

Programmeres anyone?

> And for those who fear that they will never wrap their heads around all the whys and wherefores (which a small minority has always insisted on writing as “why’s and wherefore’s) of the apostrophe, here's a cheering thought: no matter how badly you misuse this punctuation, there is a good chance that some famous writer in the past has done the same thing. Furthermore, there is a sporting chance that any mistakes you make with it will one day come back into fashion. 

So @kintel,  care to wield Solomans/Soloman's sword? Then close the issue.

closed 2026-02-28 by github-actions[bot]