openscad/openscad #2056: Object may not be a valid 2-manifold and may need repair!

kdgkp98, 2017-06-08:
[image]
Recently when I was drawing some complex models I found that sometimes polyhedrons didn't work with intersection() and/or difference(). In those case F5 and F6 would generate completely different results. Unluckily, F6 was wrong so I couldn't even export it as an STL.

For example, this code block
union(){
polyhedron( 
    points=[ [0,0,0],[109,0,0],[109,6,0],[19.13,28.46,0],[11.47,22.03,0],[9.54,24.33,0],[1.88,17.9,0],[109,6,6],[19.13,28.46,11],[109,0,6],[0,0,11],[1.88,17.9,11],[9.54,24.33,11],[11.47,22.03,11] ],
    faces=[ [0,1,2,3,4,5,6],[2,7,8,3],[10,9,7,8,13,12,11],[0,1,9,10],[0,10,11,6],[6,11,12,5],[5,12,13,4],[4,13,8,3],[1,9,7,2] ]
    );
    
    translate([109,0.5,5.5])
    rotate([0,90,0])
    polyhedron(
    points=[ [0,0,0],[1,0,27],[1,3,27],[0,5,0],[5,0,0],[4,0,27],[5,5,0],[4,3,27] ],
    faces=[ [0,1,2,3],[0,4,5,1],[4,6,7,5],[6,3,2,7],[0,4,6,3],[1,5,7,2] ]
    );
}
[image]
This is the console output after pressing F6:

Saved backup file: C:/Users/HP/Documents/OpenSCAD/backups/tria-backup-qHp13736.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 15
Geometry cache size in bytes: 94728
CGAL Polyhedrons in cache: 8
CGAL cache size in bytes: 3001008
Total rendering time: 0 hours, 0 minutes, 0 seconds
   Top level object is a 3D object:
   Simple:         no
   Vertices:        8
   Halfedges:      24
   Edges:          12
   Halffacets:     10
   Facets:          5
   Volumes:         1
WARNING: Object may not be a valid 2-manifold and may need repair! 
Rendering finished.

I am using OpenSCAD-2015.03-2 version on windows
Please help me...

MichaelAtOz, 2017-06-09:
You have some mis-ordered faces.
Preview then View/Thrown-together shows them purple:
[image]

See [Wiki (mis-ordered section)](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron)

kdgkp98, 2017-06-09:
Thanks for your reply MichaelAtOz  but we were unable to figure out how to remove those mis-ordered faces.Can you please help us??

MichaelAtOz, 2017-06-09:
I fixed the first poly, see notes. Note the layout, it helps working out what's what.

Note the `!` in the second poly, that makes it just do that bit, helps debugging one bit at a time.
So now examine the faces, make sure the points are in clockwise order.
With View/Thrown-together, the purple faces are wrong.
Once fixed remove the `!`.
```
union(){
polyhedron(
points=[ 
  [0,0,0]               //0
  ,[109,0,0]            //1
  ,[109,6,0]            //2
  ,[19.13,28.46,0]      //3
  ,[11.47,22.03,0]      //4
  ,[9.54,24.33,0]       //5
  ,[1.88,17.9,0]        //6
  ,[109,6,6]            //7
  ,[19.13,28.46,11]     //8
  ,[109,0,6]            //9
  ,[0,0,11]             //10
  ,[1.88,17.9,11]       //11
  ,[9.54,24.33,11]      //12
  ,[11.47,22.03,11]     //13
  ],
faces=[ 
  [0,1,2,3,4,5,6]       // bottom, OK
  ,[2,7,8,3]            // y+ long side, OK
  //,[10,9,7,8,13,12,11]// top, anticlockwise
  ,[11,12,13,8,7,9,10]  // correct order
  ,[0,10,9,1]           // y0 side, was anticlockwise
  ,[0,6,11,10]          // x0 side,       "
  ,[5,12,11,6]          // x0ish angle,   "
  ,[4,13,12,5]          // notch,         "
  ,[3,8,13,4]           // other angled face, "
  ,[1,9,7,2] // OK
  ]
);

translate([109,0.5,5.5])
rotate([0,90,0])
!polyhedron(
points=[ 
  [0,0,0]     //0
  ,[1,0,27]
  ,[1,3,27]   //2
  ,[0,5,0]
  ,[5,0,0]    //4
  ,[4,0,27]
  ,[5,5,0]    //6
  ,[4,3,27]
  ],
faces=[ // check these 1 is OK (yellow) other are purple
  [0,1,2,3]
  ,[0,4,5,1]
  ,[4,6,7,5]
  ,[6,3,2,7]
  ,[0,4,6,3]
  ,[1,5,7,2] 
  ]
);

}
```

kdgkp98, 2017-06-10:
Thanks for your help....

closed 2017-06-11 by MichaelAtOz