openscad/openscad #2058: From splash screen, 'New' creates blank window

acochrane, 2017-06-10:
Hey folks,

Openscad installs fine and starts fine, but when I go to open a file or make a new file, the resulting editor window is blank and the console displays some errors:

user@hostname:~$ openscad
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
composeAndFlush: makeCurrent() failed
...
repeating the composeAndFlush error until I close the window.

I get the same error with the version I compiled from source as I get from the version installed from ppa:openscad/releases.

Before I added the ppa, I tried the version from the ubuntu default repos.
It showed the text editor window but not the 3d render window.
It also didn't display any error messages on the console.

The machine this error occurs on has dual video cards, perhaps the configuration has something to do with the error?

Any hints on this?

Thanks for your time.

t-paul, 2017-06-10:
That sounds like a broken Qt (or driver issue). What's the exact Distribution version and which PPA is that? What's the output of `openscad --info` (on command line).

Maybe also try the [development snapshots](http://www.openscad.org/downloads.html#snapshots), those still use an older Qt version.

acochrane, 2017-06-10:
Ok so I just rebooted and things look to be working fine.
I think it had to do with some other video packages that I auto-removed.

Sorry for the alarm, I will close the issue.

closed 2017-06-10 by acochrane

KjellMorgenstern, 2022-08-06:
Seeing this exact issue. After the initial dialog to open a file or an example, I see a black window without any content.
Mouse clicks seem to trigger something, when I close it a dialog appears if I want to save changes.
```
user@hostname:$ ./OpenSCAD-2022.06.26.ai11958-x86_64.AppImage 
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
...
```

Using the latest dev build (but also happens with several others, like 2021.01) with `--info`:
```

 ./OpenSCAD-2022.06.26.ai11958-x86_64.AppImage --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2022.06.26.ai11958 (git ea8950bd3)
System information: Linux 5.4.0-113-generic #127-Ubuntu SMP Wed May 18 14:30:56 UTC 2022 x86_64 Ubuntu 20.04.4 LTS 12 CPUs 62.65 GB RAM
User Agent: OpenSCAD/2022.06.26.ai11958 (git ea8950bd3) (Linux x86_64; Ubuntu 20.04.4 LTS)
Compiler: GCC "7.5.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_65_1
Eigen version: 3.3.4
CGAL version, kernels: 5.4, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.2
QScintilla version: 2.10.2
InputDrivers: 
GLib version: 2.56.4
lodepng version: 20210627
libzip version: 1.1.2
fontconfig version: 2.12.6 (runtime: 2.13.1)
freetype version: 2.8.1 (runtime: 2.10.1)
harfbuzz version: 1.7.2 (runtime: 2.6.4)
cairo version: 1.15.10
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-trust-corefinement, fast-csg-debug, fast-csg-exact, fast-csg-exact-callbacks, fast-csg-remesh, fast-csg-remesh-predictibly, roof, input-driver-dbus, lazy-union, vertex-object-renderers, vertex-object-renderers-indexing, vertex-object-renderers-direct, vertex-object-renderers-prealloc, textmetrics, import-function, sort-stl
Application Path: /tmp/.mount_OpenSCX498Tv/usr/bin
<snip>
...
directories and font files here
...
</snip>
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  152 (GLX)
  Minor opcode of failed request:  24 (X_GLXCreateNewContext)
  Value in failed request:  0x0
  Serial number of failed request:  34
  Current serial number in output stream:  35
```

All the older OpenScad AppImages have been working before, on the same system, no major config changes. So indeed the issue is very likely to go away after rebooting, but I'll try a bit more first.